Ultimaker/Cura #5540: Top/Bottom thickness (mm) entirly broken

dgrat, 2019-03-31:
Still ignoring the values

[image]

smartavionics, 2019-03-31:
For me, using my master branch based Cura, changing either the Layer Height or the Top/Bottom Thickness settings changes the values of Top Layers and Bottom Layers. I can also set the number of layers explicitly but don't need to when those other settings are changed. Of course, if you do set the value of Top Layers or Bottom Layers explicitly, then you would not expect changes to Layer Height or Top/Bottom Thickness to have any effect.

Ghostkeeper, 2019-04-01:
Going to assume your reproduce steps here because you didn't write them down. I'm assuming that you changed the Bottom Layers setting, then changed the Top/Bottom Thickness setting and expect the Bottom Layers setting to change.

That's the wrong expectation. Bottom Layers overrides Top/Bottom Thickness.

closed 2019-04-01 by Ghostkeeper