Ultimaker/Cura #5542: [4.0.0] Text Font corrupted on Neon (Ubuntu + KDE)

alvaro893, 2019-03-31:
**Application Version**
4.0.0

**Platform**
KDE Neon 5.15
Ubuntu 18.04 is the base system

**Printer**
Anycubic i3 mega

**Steps to Reproduce**
It is quite inconsistent. Randomly happens when using with the UI. Sometimes it doesn't happen at all.

**Actual Results**
Text Font gets corrupted

**Expected results**
Unmodified text

**Additional Information**
[image]
[image]

droidmahn, 2019-03-31:
I'm having similar issues Arch linux + KDE

nallath, 2019-04-01:
Are you using the appimage?

alvaro893, 2019-04-01:
@nallath  yes.

Ghostkeeper, 2019-04-01:
Haven't seen this happen, so far. Neither on the AppImage or running from source on my Ubuntu 18.10 computer.

I'm not quite sure how to approach a fix for this. Do you know if this happened also in 3.6 or before?

alvaro893, 2019-04-02:
I downloaded the 3.6 AppImage to test it. It seems that it works just fine. No text corruption. Could this be related somehow to Nvidia drivers?

Ghostkeeper, 2019-04-03:
Maybe. Hard to say unless we find a pattern in the people who experience this. I'm not seeing this on the NVidia drivers on Windows but that's a whole different beast of course.

It's funny how the Prepare, Preview and Monitor stages are still rendering fine.

alvaro893, 2019-04-03:
I tried 4.0.0 AppImage in a old laptop with Kubuntu installed on it. It worked fine. This laptop uses Intel Integrated Graphics (rev 02) instead.

Ghostkeeper, 2019-04-04:
Maybe it only happens for text that is not exactly pixel-aligned?

fieldOfView, 2019-04-04:
It looks like it is a specific font size. That could mean the bitmap representation that is created when a font is loaded at a certain size is corrupted somehow.

The text looks well aligned to me; I checked the rendering of those element, and non-aligned text shows up on Windows because it uses a different renderer.

alvaro893, 2019-04-11:
Some news about this issue!

After going to Settings->Fonts->Anti-aliasing and set it from "vendor default" to "enabled". The font corruption disappear altogether.

Also fonts look a bit better.

bmveee, 2019-08-08:
One more note to add to this - it's not enough to 'enable' anti-aliasing, you have to go to Configure (from the Font menu) and set it to Full hints. At least that's what I had to do...

closed 2020-06-19 by no-response[bot]