Ultimaker/Cura #5545: [master] Question about initialisation of extruder_prime_layer_nr

smartavionics, 2019-04-01:
I'm investigating why a dual extruder print is including a spurious prime travel move and I think it's because the value of extruder_prime_layer_nr for the layer in question is wrong. The extraneous travel occurs when switching from extruder 2 to extruder 1 on the second layer. On both layers 1 and 2, both extruders are used so by the time layer 2 is being printed, I would have thought that neither extruders would need priming.

Looking at how extruder_primer_layer_nr gets initialised, it seems to me that for the first used extruder on layer 0, extruder_prime_layer_nr does not get a value of 0 and it will get a value depending on the lowest layer (above layer  0) that it next used.

Surely, extruder_prime_layer_nr for the first extruder used on layer 0 should be set to 0?

Could a dev please explain why with the current code is how it is?

This shows the spurious 10mm travel move inserted at the nozzle switch on layer 2.

[image]

Ghostkeeper, 2019-04-01:
I would expect `extruder_prime_layer_nr[extruder_nr]` to get a value of `0` if they are used for that extruder number on layer 0. That's what is written in the code as I read it:

https://github.com/Ultimaker/CuraEngine/blob/c1c8bcb550805aae8fa091a2581ddef1e55e1244/src/FffGcodeWriter.cpp#L1030

smartavionics, 2019-04-01:
But the 2nd extruder used on the 2nd layer is the 1st extruder used on the 1st layer so its entry in extruder_prime_layer_nr is going to be > 0.

I don't understand why extruder_prime_layer_nr isn't assigned for all the extruders used on the layer rather than just the last. What happens if you have a printer with more than 2 extruders? (I realise that isn't an issue for Ultimaker but Cura does support > 2 extruders).

Ghostkeeper, 2019-04-01:
Hmm, yeah I wonder why it's only registering the last extruder on a layer.

smartavionics, 2019-04-01:
Just trialing this change, cures the immediate problem (unwanted prime travel at start of 2nd layer) but I don't have enough knowledge and experience of multi-material printing to really be sure it's right.
````
diff --git a/src/FffGcodeWriter.cpp b/src/FffGcodeWriter.cpp
index 19b32888..e17f81f5 100644
--- a/src/FffGcodeWriter.cpp
+++ b/src/FffGcodeWriter.cpp
@@ -1023,7 +1023,14 @@ void FffGcodeWriter::calculateExtruderOrderPerLayer(const SliceDataStorage& stor
         std::vector<std::vector<size_t>>& extruder_order_per_layer_here = (layer_nr < 0) ? extruder_order_per_layer_negative_layers : extruder_order_per_layer;
         extruder_order_per_layer_here.push_back(getUsedExtrudersOnLayerExcludingStartingExtruder(storage, last_extruder, layer_nr));
         last_extruder = extruder_order_per_layer_here.back().back();
-        extruder_prime_layer_nr[last_extruder] = std::min(extruder_prime_layer_nr[last_extruder], layer_nr);
+        std::vector<bool> extruder_is_used_on_this_layer = storage.getExtrudersUsed(layer_nr);
+        for (size_t extruder_nr = 0; extruder_nr < extruder_is_used_on_this_layer.size(); ++extruder_nr)
+        {
+            if (extruder_is_used_on_this_layer[extruder_nr])
+            {
+                extruder_prime_layer_nr[extruder_nr] = std::min(extruder_prime_layer_nr[extruder_nr], layer_nr);
+            }
+        }
     }
 }
````

smartavionics, 2019-04-02:
Hi @Ghostkeeper , if you think the above suggested diff looks plausible, I'll put it in a PR.

Ghostkeeper, 2019-04-02:
We may still prime even if you do have a brim. In fact we generate a skirt for every used extruder if you have a brim and then still do the prime blob.

smartavionics, 2019-04-02:
So where does the prime occur for the 2nd (and subsequent) extruders?

OK, what's really confusing me here is that planPrime() does a 10mm move to the N irrespective of whether the prime blob is enabled or not. Is that  where the bug is? Should it only do that 10mm move if the prime blob is actually output?

Ghostkeeper, 2019-04-02:
In the prime blob location of that second extruder (it is a per-extruder setting).

We recently had a bug about moving to the prime blob location even if the prime blob was disabled, but I thought we fixed that?

smartavionics, 2019-04-02:
> In the prime blob location of that second extruder (it is a per-extruder setting).

But the prime doesn't occur until the first layer that extruder is used on. So could do the prime on layer 100 (or whatever) is that right?

> We recently had a bug about moving to the prime blob location even if the prime blob was disabled, but I thought we fixed that?

The way the code looks right now, it omits the travel to the blob location if the prime is disabled but still calls planPrime() irrespective of whether the prime blob is disabled.

smartavionics, 2019-04-02:
So perhaps the change should embody this:
````
@@ -3324,9 +3328,8 @@ void FffGcodeWriter::setExtruder_addPrime(const SliceDataStorage& storage, Layer
                 bool prime_pos_is_abs = train.settings.get<bool>("extruder_prime_pos_abs");
                 Point prime_pos = Point(train.settings.get<coord_t>("extruder_prime_pos_x"), train.settings.get<coord_t>("extruder_prime_pos_y"));
                 gcode_layer.addTravel(prime_pos_is_abs ? prime_pos : gcode_layer.getLastPlannedPositionOrStartingPosition() + prime_pos);
+                gcode_layer.planPrime();
             }
-
-            gcode_layer.planPrime();
         }
 
````

Ghostkeeper, 2019-04-03:
Perhaps. Before we make the change though, can you give an example of where it goes wrong so that we can test whether it is really wrong and whether it is then fixed?

smartavionics, 2019-04-03:
I attach the project file that has been supplied to me. With stock Cura, you get an unwanted prime travel when switching between extruders on layer 2.

[CUP_Test_Cura_master.zip](https://github.com/Ultimaker/Cura/files/3037726/CUP_Test_Cura_master.zip)

smartavionics, 2019-04-03:
Here's the printer definition that you need for that project...

[Mark2_definition.zip](https://github.com/Ultimaker/Cura/files/3037753/Mark2_definition.zip)

smartavionics, 2019-04-03:
Just to recap, I'm currently advocating these changes...

````
diff --git a/src/FffGcodeWriter.cpp b/src/FffGcodeWriter.cpp
index 19b32888..1fd3676e 100644
--- a/src/FffGcodeWriter.cpp
+++ b/src/FffGcodeWriter.cpp
@@ -1023,7 +1023,11 @@ void FffGcodeWriter::calculateExtruderOrderPerLayer(const SliceDataStorage& stor
         std::vector<std::vector<size_t>>& extruder_order_per_layer_here = (layer_nr < 0) ? extruder_order_per_layer_negative_layers : extruder_order_per_layer;
         extruder_order_per_layer_here.push_back(getUsedExtrudersOnLayerExcludingStartingExtruder(storage, last_extruder, layer_nr));
         last_extruder = extruder_order_per_layer_here.back().back();
-        extruder_prime_layer_nr[last_extruder] = std::min(extruder_prime_layer_nr[last_extruder], layer_nr);
+
+        for (size_t extruder_nr : extruder_order_per_layer_here.back())
+        {
+            extruder_prime_layer_nr[extruder_nr] = std::min(extruder_prime_layer_nr[extruder_nr], layer_nr);
+        }
     }
 }
 
@@ -3324,9 +3328,8 @@ void FffGcodeWriter::setExtruder_addPrime(const SliceDataStorage& storage, Layer
                 bool prime_pos_is_abs = train.settings.get<bool>("extruder_prime_pos_abs");
                 Point prime_pos = Point(train.settings.get<coord_t>("extruder_prime_pos_x"), train.settings.get<coord_t>("extruder_prime_pos_y"));
                 gcode_layer.addTravel(prime_pos_is_abs ? prime_pos : gcode_layer.getLastPlannedPositionOrStartingPosition() + prime_pos);
+                gcode_layer.planPrime();
             }
-
-            gcode_layer.planPrime();
         }
 
         if (gcode_layer.getLayerNr() == 0 && !gcode_layer.getSkirtBrimIsPlanned(extruder_nr))
````

Ghostkeeper, 2019-04-15:
We're not seeing this issue anywhere else than in this project. We don't consider this to be a big problem so far, this one extra travel move, even if it might hide some other issue. When the hidden becomes visible, we'll consider debugging it further.

smartavionics, 2019-04-15:
So should I put the above diff into a PR or not bother?
There's two parts to the diff, the first part is possibly a separate issue in that the original code only updates extruder_prime_layer_nr for the last extruder used on each layer and not for all extruders used on each layer. Surely that needs changing?

As for the moving of the call to planPrime(), that would fix the extraneous travel move. Personally, I don't care as I only use single extruder printers.

smartavionics, 2020-07-14:
Most likely now fixed. Closing...

closed 2020-07-14 by smartavionics

Ghostkeeper, 2020-07-15:
Yeah should be fixed by your PR. Thanks for investigating!