Ultimaker/Cura #5546: [4.0] Octoprint camera picture in Monitor window not showing with reverse proxy that requires http auth with username and password

sjbotha, 2019-04-01:
**Application Version**
4.0 and 3.6

**Platform**
Windows 10

**Printer**
Tevo Tornado

**Steps to Reproduce**
Run Cura and click Monitor

**Actual Results**
No webcam image

**Expected results**
The webcam image is shown

**Additional Information**
I'm using Cura 4.0 right now. I also had this issue with Cura 3.6.

My octoprint is accessible at /octoprint path and webcam is at /octoprint/webcam. After configuring the reverse proxy to not require authentication for /octoprint/webcam the image showed up.

I made logs of the camera not working and then working. I did not see a difference in the log.

[cura webcam not working.log](https://github.com/Ultimaker/Cura/files/3029642/cura.webcam.not.working.log)
[cura webcam working.log](https://github.com/Ultimaker/Cura/files/3029643/cura.webcam.working.log)

fieldOfView, 2019-04-01:
Please report issues with the OctoPrint plugin in the OctoPrint plugin issue queue:
https://github.com/fieldofview/Cura-OctoPrintPlugin/issues.

sjbotha, 2019-04-01:
Created issue there.

closed 2019-04-01 by sjbotha

fieldOfView, 2019-04-01:
Thanks, see you there.