Ultimaker/Cura #5547: [4.0] Printer settings plugin doesn't work

Liger0, 2019-04-01:
**Application Version**
4.0

**Platform**
Windows 10 x64

**Printer**
Custom Double extrusion cr-10s

**Steps to Reproduce**
Download and install from the market the "printer settings" plugin, then restart Cura.

**Actual Results**
Nothing happens.

**Expected results**
Like on the older versions, it should allow to see the hidden machine settings.

fieldOfView, 2019-04-01:
What do you expect to happen?
The plugin adds settings to Cura. It does not mean they are visible by default. Have you tried using the "Search" field on top of the Custom setting to search for eg "nozzle"? That should show you the nozzle size setting, which is part of Printer Settings.

Liger0, 2019-04-01:
You're right, I was sure the visibility was set to "All", but it seems after adding the plugin that step should be done again.
I think this could be improved so the "Set all visible" applies even after new options are added after plugins though.

closed 2019-04-01 by nallath

fieldOfView, 2019-04-01:
Also, for the record:
https://github.com/fieldofview/cura-printersettingsplugin/issues

fieldOfView, 2019-04-01:
> I think this could be improved so the "Set all visible" applies even after new options are added after plugins though.

I don't think we should further encourage people making all settings visible by default. That practice causes many performance and UX issues. You really, really don't need all those settings. Just make the ones you actually use visible, and use the search for the rest.