Ultimaker/Cura #5549: [Feature request] Profile name preset

Liger0, 2019-04-01:
Hi there.
I always wondered why when exporting a profile, it doesn't use as a preset the profile name inside of Cura, and instead it just uses a blank text in the box, and if you have multiple profiles it could lead to user confusion when writing it each time.
Would it be possible to implement this?

fieldOfView, 2019-04-01:
Please fill in the issue template. It may be tedious for you, but now it is tedious for us trying to find out what you mean.

Liger0, 2019-04-01:
> 
> 
> Please fill in the issue template. It may be tedious for you, but now it is tedious for us trying to find out what you mean.

I don't think it's an issue, as stated it's a feature request.

Ghostkeeper, 2019-04-02:
Yeah, small ease of use thing that could be improved.

nallath, 2019-04-03:
Unfortunately, the way this file dialog is created means that it's impossible to set this. If we do want to be able to set this, we need to do quite some changes. As it's not that high of a priority, we will postpone this for the moment.