Ultimaker/Cura #5550: [4.0.0] Layer view: incorrect material color

Block137, 2019-04-01:
**Application Version**
4.0

**Platform**
Windows10

**Steps to Reproduce**
1. Use multi-extruder setup
2. Configure a print that start printing with other tools than extruder1

**Actual Results**
Material color of the part printed with extruder2 is shown in the color of extruder1
But the generated G-code has correct extruder output.

**Expected results**
Show correct material color

**Additional Information**
Pic: Brim extruder is set to green (extruder2), but the preview showed yellow (extruder1)
G-code file started with T1 (no problem)
[image]

Ghostkeeper, 2019-04-02:
Cannot reproduce this issue :(

[image]

Block137, 2019-04-02:
Set the "Build plate adhesion extruder" to extruder2
The brim color should remain yellow (supposed to turn green)

That's what I think is missing

Ghostkeeper, 2019-04-02:
Ah, you're right. It reproduces then. Seems to be specific to the build plate adhesion.

Cura doesn't take the custom starting g-code in the preview, no, unless you save the g-code and re-load it again (but then you lose the material colours altogether iirc).

I think this is caused by CuraEngine. It sends the g-code separately from the layer preview, because the front-end doesn't have the computational power to calculate the extrusion widths in time. Perhaps we forgot to set the extruder of the build plate adhesion in the preview data.

Block137, 2019-04-06:
@Ghostkeeper I also just found that Cura won't set temperature for build plate adhesion extruder.

nallath, 2019-04-08:
For our reference; CURA-6445

nallath, 2019-05-03:
This has been fixed for 4.1

closed 2019-05-03 by nallath