Ultimaker/Cura #5551: [4.0] Error message on startup, then crash.

ryandavidsongns, 2019-04-01:
Hey,

I've been using 4.0 since it was released but this morning when i tried to open it i get the error in the screenshot. Once i close the error, cura also closes. I believe there were some microsoft updates rolled out on our work PCs over the weekend, so this may have something to do with it?
Also 3.6 works fine.

[image]
[DxDiag.txt](https://github.com/Ultimaker/Cura/files/3031084/DxDiag.txt)
[cura.log](https://github.com/Ultimaker/Cura/files/3031085/cura.log)

ryandavidsongns, 2019-04-01:
> It looks like a file in your Cura configuration got corrupted in an interesting way.
> 
> If you go to C:\Users\ryand\AppData\Roaming\cura\ and rename the folder 4.0 to something like 4.0.bak, you should be able to start Cura again.

Thanks, that sorted it!

fieldOfView, 2019-04-01:
Good to hear. If you could zip up that old 4.0 folder and post a link to it, we can see what actually went wrong (and hopefully get at least a more useful crash dialog that helps you reset Cura the way you now did manually).

ryandavidsongns, 2019-04-01:
> Good to hear. If you could zip up that old 4.0 folder and post a link to it, we can see what actually went wrong (and hopefully get at least a more useful crash dialog that helps you reset Cura the way you now did manually).

Here you go

ryandavidsongns, 2019-04-01:
> Good to hear. If you could zip up that old 4.0 folder and post a link to it, we can see what actually went wrong (and hopefully get at least a more useful crash dialog that helps you reset Cura the way you now did manually).

Here you go
[4.0.bak.zip](https://github.com/Ultimaker/Cura/files/3031540/4.0.bak.zip)

fieldOfView, 2019-04-02:
DEVS: in GlobalStack.configuredConnectionTypes, there is no resilience against a None in the list of connectiontypes. Needs a try/except.

nallath, 2019-04-02:
Good spot. Fixed that for 4.1

commit 5e66564c (2019-04-02): Ensure that configurationTypes doesn't crash if a None is in the list

Fixes #5551
closed 2019-04-02 by nallath