Ultimaker/Cura #5552: [4.0.0] UI not visible (wayland, sway, nouveau)

tobinjones, 2019-04-01:
**Application Version**
Cura 4.0.0

**Platform**
Arch linux, up-to-date (5.0.5-arch1-1-ARCH). Running sway window manager on Wayland. NVIDIA graphics card running Nouveau driver.

**Steps to Reproduce**
1) Run `cura`

**Actual Results**
Splash screen shows, but Cura window is garbled mess.

**Expected results**
Cura should show!

**Additional Information**
- I get the output `Using Wayland-EGL` when cura is started.
- Screenshot of effect: [image]
- Output of `cura --debug &> out.txt`: [out.txt](https://github.com/Ultimaker/Cura/files/3031512/out.txt)
- Note, `glxgears` works fine.

nallath, 2019-04-02:
You're not using the appimage, which is the only version that we support. There is at least one package missing from the version of Cura that was packaged by whoever made the package you are using.

tobinjones, 2019-04-02:
Thanks for that, I tried the appimage, and it failed in the same way.

I've now got things working, it turns out it was my fault -- I had the environment variable `QT_QPA_PLATFORM=wayland_egl` set, which forced cura to use wayland rather than x11 via xwayland.

Deleting that environment variable, or setting `QT_QPA_PLATFORM=xcb` fixes things. Whoops! Cheers!

closed 2019-04-02 by tobinjones