Ultimaker/Cura #5553: [4.0] "Infill layer thickness" -setting wont become visible

swatti87, 2019-04-02:
Even if i tick ALL the boxes some settings wont become visible.

This is a VERY important setting for me as i do my infilling every 3-4 layers to save time.

fieldOfView, 2019-04-02:
Make sure that the 'infill density' is greater than 0, and 'spaghetti infill' is not enabled.

Ghostkeeper, 2019-04-02:
Not able to reproduce, unless I set Infill Density to 0 or enable Spaghetti Infill. Seems to be behaving as expected. What are your reproduce steps?

Maybe there is a bug that this setting gets shown too often if you set the infill line distance to 0. It shouldn't show up then I suppose.

swatti87, 2019-04-05:
Hmm... It may be due to the fact that i used a different-than my layer height in 3.6

Is there a way to just "force" all settings to be visible when toggled? I fully prepared to face all mess that might introduce.

Ghostkeeper, 2019-04-11:
If your infill layer thickness is not an exact multiple of the layer height it will get rounded.

There is a mod that you could make to make all settings visible even if they don't have an effect on your print: Go into C:\Program Files\Ultimaker Cura 4.0\resources\definitions\fdmprinter.def.json and remove all lines there with `"enabled":` in them. However that will make settings visible of which Cura thinks they are not used in your print, so if you then edit the setting nothing will happen. Of course, be sure to also use the hamburger menu next to the settings search bar to set "Show All Settings".

closed 2020-06-20 by no-response[bot]