Ultimaker/Cura #5554: Super-fat infill lines leaves gaps, how to fill those?

swatti87, 2019-04-02:
[image]

I do all my prints 100% infilled. Meaning the infilling is the most time consuming part of most prints.

It turns out my nozzle can do quite a lot fatter lines than one would imagine, BUT cura leaves gaping holes where it thinks the infill-lines dont fit. They would, in most cases if they were single-lines

Is there a way to fill those holes with some other form of infill? Skin/top/bottom? Is there a way to tell the system to lower the height and width of the infill-line "where needed" ?

Ghostkeeper, 2019-04-02:
The "Fill Gaps Between Walls" setting should fill those up, if enabled. Is it enabled in your settings?

swatti87, 2019-04-02:
Ookkey... Maybe solved.

"infill overlap" If you have, a say 1.2mm nozzle and set this to 0, the layer-preview will show the infill overlap the walls by a LOT. However, it would seem,  and i cant confirm this easily that the actual line does end in the right spot, the line is just drawn "wrong" in the preview. 

Fairly easy to test by using 2mm "infill line-width" and "infill overlap" at something realistic like 0.2-0.3mm
The preview overlaps over the wall but the actual print seemed as if it was correct...

Line width at 1mm and height at 0.8 even the mighty Bondtech started to grind. PETG viscosity at 270c 
 and 70mm/s had been reached. 

It appears the gaps were made by the overlap-setting as i was trying to make the preview look normal.
I toned down the settings a bit and second print is running.

digitalfrost, 2022-07-27:
**Stale ticket** from 2019.
Looks like @swatti87 resolved his issue.
Can be closed.

jellespijker, 2022-07-27:
Agreed

closed 2022-07-27 by jellespijker