Ultimaker/Cura #5555: [4.0.0] Arrange all models CTRL+R

Martin-S-2019, 2019-04-02:
**Application Version**
4.0.0

**Platform**
Win 7 Pro

**Printer**
Ultimaker 3

**Steps to Reproduce**
Arrange all models CTRL+R

**Actual Results**
Wrong postions

**Expected results**
A better arrangement

**Additional Information**

[image]

[image]

[image]

[image]

[Cura.zip](https://github.com/Ultimaker/Cura/files/3033024/Cura.zip)

nallath, 2019-04-02:
This is because the arrangement algorithm is pretty damn naive. It currently tries to put the largest object in the middle and then put all the other objects around it. That logic tends to work okayish if you have one large model and a bunch of smaller ones but it tends to break down fast if that's not the case.

We still have a ticket on the backlog to spend time on it and it's also one of the areas where I still want to get some (master) graduate(s) for to do some more research on, but I don't think this will be fixed any time soon.

closed 2019-04-02 by nallath

supernovae, 2019-04-15:
Just confirming i'm seeing this same behavior. It did work for the most part in cura 3.x but i also noticed when i import a couple of models in cura 4 that the models import off the virtual build plate after the first one is imported. Have to change my view and drag them back into build plate.