Ultimaker/Cura #5557: [4.0.0] output overview not showing correctly

moham96, 2019-04-02:
**Application Version**
4.0.0

**Platform**
Manjaro linux(Arch linux based distro) running i3wm under X.org

**Printer**
Creality Ender 3

**Steps to Reproduce**
1- Load an STL file.
2- Adjust the settings and click "Slice".
3- Hover the mouse over the info icon in the right bottom corner.
4- You will see that the pop doesn't show the information correctly.

**Actual Results**
The output overview popup doesn't show correctly (the text shows outside the tooltip)

**Expected results**
The text should show inside the tooltip

**Additional Information**
Screenshot of the problem:
[image]

Ghostkeeper, 2019-04-02:
Duplicate of https://github.com/Ultimaker/Cura/issues/5307.

The issue is caused by using Qt 5.12 which is not backwards compatible with Qt 5.10. The official Cura distribution is using Qt 5.10. You can get Cura to work by using our AppImage distribution.

closed 2019-04-02 by Ghostkeeper

moham96, 2019-04-02:
@Ghostkeeper Ok, thanks will use the AppImage for now.

moham96, 2019-04-07:
another issue I noticed that might also be cause by qt version is that the "sidebar gui" extension doesn't work with archlinux cura but it does work with AppImage version