Ultimaker/Cura #5558: cura4.0 Extuder stops

Den54, 2019-04-02:
Hi
 The Problem I seem to have is that once the print job has started ,All goes well for the first 
20 to 30 layers and then the extruder stops. But the print/hot end is still going through the 
motion for the model. I get the same problem printing from USB or if I save to Gcode.
 It's the same no matter what model I use. I have tried the Gcode in a different slicer , but still does not print. Also I have loaded a pre-sliced Gcode in to Cura and it worked. So I am guessing its something to do with the slicing?. Any help would be greatly appreciated. Thanks...  
 
**_All files are in Zip folder.._**

**Application Version**
Cura 4.0 or 3.6

**Platform**
Win 10

**Printer**
Ender 3

**Steps to Reproduce**
Load model, slice and Print

**Actual Results**
After print has started around Layer 20 to 30 Extruder stops.

**Expected results**
whole model to be printed

**Additional Information**

[CURA files.zip](https://github.com/Ultimaker/Cura/files/3034243/CURA.files.zip)

smartavionics, 2019-04-02:
I think this has been a problem before, try turning off the acceleration and jerk control in the slicer.

Ghostkeeper, 2019-04-02:
The g-code looks okay. I think Cura does what it's supposed to do.

Den54, 2019-04-02:
Hi
Thanks for your quick response :) . I did not enable acceleration or jerk settings. It was a fresh install, so just set basic settings. Gcode does look ok. But print failed when loaded into different slicer. This seem so odd. Thanks for the help.  I have not had any problems printing same models in a different slicer.

smartavionics, 2019-04-02:
But your gcode has the accel/jerk control enabled, so it may be having an effect.

Den54, 2019-04-02:
Ok.  smartavionics , You are Right lol. I had to enabled Acceleration and Jerk in settings and disabled them both lol that's strange . But all is working now as should be . Thank you so much for your help and I will now close the issue. Thanks :))

closed 2019-04-02 by Den54