Ultimaker/Cura #5559: [4.0] Tree supports have no infill

Liger0, 2019-04-02:
**Application Version**
4.0 (also old versions)

**Platform**
Windows 10 x64

**Printer**
Dual estruder cr-10s

**Steps to Reproduce**
Load the .3mf, use tree supports

**Actual Results**
The tree supports have no infill so branches try to start in the air and fail. It randomly happened also in all the older versions of Cura I tried.

**Expected results**
The tree supports should have infill. 

**Additional Information**
[image]
[tree support infill.zip](https://github.com/Ultimaker/Cura/files/3036231/tree.support.infill.zip)

closed 2019-04-02 by Liger0

Ghostkeeper, 2019-04-03:
People experiencing this as well: The default setting for tree support is to set the Support Infill Density to 0, but it can still be changed if you want it to have infill.

Liger0, 2019-04-03:
> 
> 
> People experiencing this as well: The default setting for tree support is to set the Support Infill Density to 0, but it can still be changed if you want it to have infill.

Yeah, I noticed that after posting it. The weird thing to me is that it takes the same box of the standard support "infill density", but they are indipendent. So when you enable tree support, the value in the box changes automatically and I didn't notice it. That's A LOT confusing, and I think the 2 box should be kept separate.

Ghostkeeper, 2019-04-04:
Yeah one of the requirements of Tree Support to not be experimental any more is to merge the setting with normal support enable, and just have a drop-down of the support strategy to use.