Ultimaker/Cura #5560: models end up below the build plate

marijnd, 2019-04-03:
When I open attached 3mf file, wich consists of multiple models, a large number of the models end up below the build plate.
They were correctly positioned on top of the build plate when I saved the project file.

**Application Version**
4.0.0

**Platform**
Linux (Ubuntu 18.04)

**Printer**
UM3

**Steps to Reproduce**
Open the 3mf project file, look at how the models are placed.

**Actual Results**
A large portion of the models end up below the build-plate

**Expected results**
The models would be positioned exactly as they were when I saved them, on top of the build plate

**Additional Information**
[image]

[UM3_Lizard.3mf.zip](https://github.com/Ultimaker/Cura/files/3037289/UM3_Lizard.3mf.zip)

nallath, 2019-04-03:
I've done a bit of investigation into this and there is something weird going on with this model. The original STL (https://www.thingiverse.com/thing:3505006) is not watertight. After subdividing it with the mesh tweaker plugin it added a bunch of weird models. 

I'm still not quite sure what causes it, but my best bet is some perfect storm between project loading / saving and the mesh tools plugin (so tag @fieldOfView). 

Regardless of what caused it, it's pretty weird that the project file doesn't fully restore the state of the buildplate when loading it again, so we need to look into that.

fieldOfView, 2019-04-04:
> After subdividing it with the Mesh Tools plugin it added a bunch of weird models.

Confirmed that there seem to be "empty" models in there. I'll see if I can detect those when splitting so they don't get added to the scene.

nallath, 2019-04-04:
I've did a bit more poking around and it seems to only happen if the mesh is not watertight. 

So this gives the following reproduction steps:
1. Load a mesh that has sub parts and is not watertight
2. Use split model into parts
3. Save as .3mf
4. load .3mf file
5. Objects are not in the same spot.

Weirdly enough, all of this doesn't cause issues if all the models are watertight.

fieldOfView, 2019-04-04:
Does that mean you have a simpler model that reproduces the issue? Loading that lizard takes annoyingly long.

nallath, 2019-04-04:
Yup. I just loaded some models in Cura and saved them as a single STL. 
[Uploading MergedThing.zip…]()

GregValiant, 2024-11-15:
Is this still an issue with current Cura versions (5.8.0 and up).  Can this be closed?

closed 2024-12-10 by github-actions[bot]