Ultimaker/Cura #5564: Setting up materials for 8 extruder custom machine fails on material 8

MB3Y, 2019-04-03:
**Application Version**
4.0.0
**Platform**
Windos
**Printer**
Custom 8 extruder machine
**Steps to Reproduce**
Define an 8 extruder machine and set the materials per extruder.
All fine up till extruder 7 - drop downbox available.
**Actual Results**
For extruder 8 you can only select a very limited amount of materials.
**Expected results**
I expect to have the same possibilities for selection in extruder 8
**Additional Information**
In the attached pictures the Head and all next sections are selected as extruder 8-7-6-5-4-3 (as a good observer can see of course due to the colors matched to them 👍 )
[image]
[image]

Ghostkeeper, 2019-04-04:
I can't reproduce this, but I have a strong feeling that your 8th extruder is not configured the same way as the rest. So please do these steps:

1. Click your printer's name and click on "Manage printers".
2. Click the "Machine Settings" button.
3. Go to the tab for extruder 8.
4. Check whether the "Compatible Material Diameter" field is properly set to 1.75mm.

MB3Y, 2019-04-04:
And - yes - we got a winner!!
Filament diameter was indeed still on the default instead of the 1.75mm like all the others!
Thanks for figuring that out. You can close this point.

closed 2019-04-05 by nallath