Ultimaker/Cura #5565: [4.0] Custom machine renaming weird behaviour

Liger0, 2019-04-03:
**Application Version**
4.0

**Platform**
Win10 x64

**Printer**
Custom

**Steps to Reproduce**
I am unsure..

**Actual Results**
When I try to remove the number of printer I never written, nothing happens. If I remvoe part of the machine name, the number goes off, but if I write the letter again, the number reappears, it seems like the machine name under number 1 and 2 seem to be there but invisible.

**Expected results**
It should permanently delete invisible mahcines, I guess?

**Additional Information**
[Video](https://youtu.be/W0w8Ad4h7NY)

Ghostkeeper, 2019-04-04:
We have a few technical limitations here that make this a bit difficult to fix. There's profiles underwater that share the same unique profile names, which is causing this rename.

Given this complexity, I think we'll have to defer this to later. Maybe we can eventually make only the IDs unique and not the names, or unlink the two better.

Liger0, 2019-04-08:
They annoying thing is I keep receiving popups to reset Cura because of damaged profiles and materials even though they don't exist anymore.

Ghostkeeper, 2019-04-12:
That makes this a different issue though. How did you arrive at damaged profiles?

We won't delete damaged profiles outright because that can cause a chain reaction of profiles being deleted even if they are perfectly fine. Working with a broken configuration is a dangerous thing and we'd rather not delete our users' data. That said, broken configurations shouldn't happen unless you modified the configuration folder by hand at some point.

Liger0, 2019-04-12:
> 
> 
> That makes this a different issue though. How did you arrive at damaged profiles?
> 
> We won't delete damaged profiles outright because that can cause a chain reaction of profiles being deleted even if they are perfectly fine. Working with a broken configuration is a dangerous thing and we'd rather not delete our users' data. That said, broken configurations shouldn't happen unless you modified the configuration folder by hand at some point and made a mistake there.

I never modified the configuration folder. I just created, modified, duplicated and deleted custom machines done by Cura and someway this always happened. So not only the deleted ones weren't really deleted, but they resulted damaged aver deleting them.
I don't know precisely how to reproduce it to be honest, but I didn't do anything else.

closed 2023-06-24 by github-actions[bot]