Ultimaker/Cura #5566: Drop down menus do not work

rjmaker, 2019-04-03:
**Application Version**
4.0.0 & 3.6.0
**Platform**
Windows 10
**Printer**
Creality Ender 3
**Steps to Reproduce**
Attempt to use any drop down menu
**Actual Results**
Drop downs briefly show contained text then fill completely with Gray, impossible to tell what each item is or how many items are on the menu.
I have tried a fresh install, issue still occurs.
**Additional Information**
Same issue experienced with 3.6.0 but I ignored it.
[image]

Ghostkeeper, 2019-04-04:
Doesn't reproduce for me on Ubuntu, but this sort of thing tends to be very OS-specific.

Does it also happen for the drop-downs that allow you to select a material?

Ghostkeeper, 2019-04-04:
Maybe we can see if other people are seeing this as well and hope that we can find a pattern in the people who see this bug.

rjmaker, 2019-04-04:
Thankyou @Ghostkeeper for adding labels, I'm new to this :)

**Additional info**
This issue effects the drop down for material select, Print profile, right clicking anywhere on the screen and the search settings menu **BUT** not the drop downs such as infill type, build plate adhesion e.t.c for clarity I have attached pictures. [image]

Ghostkeeper, 2019-04-05:
Yeah the drop-downs that you see working fine are custom implementations we made, rather than system-styled drop-downs.

nallath, 2019-04-08:
The last time we had issues like that, the graphics drivers caused it (a very specific version in that case. So either upgrading or downgrading them resolved the issue)

closed 2020-06-20 by no-response[bot]