Ultimaker/Cura #5568: Cura 4.0.0 slicing hangs at 50%

ericduns, 2019-04-03:
**Application Version**
4.0.0

**Platform**
Windows 10 Enterprise

**Printer**

Creality Ender 3

**Steps to Reproduce**

Added an .stl file and clicked slice

**Actual Results**

slices to about 50%, after a while backs up to about 30% and jumps back up to 50% and then stays hung.

**Expected results**

slicing hangs at 50%

**Additional Information**

[curaLogFile.zip](https://github.com/Ultimaker/Cura/files/3040962/curaLogFile.zip)

[CE3_Horn.curaproject.3mf.zip](https://github.com/Ultimaker/Cura/files/3040966/CE3_Horn.curaproject.3mf.zip)

[image]

closed 2019-04-03 by ericduns

Ghostkeeper, 2019-04-04:
Did you find a solution to this? Why was it closed?

ericduns, 2019-04-04:
Yes, turned out that having tree supports enabled on this project seemed to
make it hang. After switching back to default supports it sliced just fine.

Ghostkeeper, 2019-04-05:
For some highly detailed and tall models, tree support can take a significant amount of time. As in, half an hour is not entirely unusual.

ericduns, 2019-04-05:
Oh wow thanks for the info!