Ultimaker/Cura #5569: [3.6 and 4.0] Cura doesn't start on Windows 10 Pro 64bit unless Windows 8 Compatibility is selected

reibuehl, 2019-04-04:
**Application Version**
3.6 and 4.0

**Platform**
Windows 10 Professional 64-bit, Version 1809

**Printer**
Ultimaker 2

**Steps to Reproduce**
After installing Windows 10 Version 1809, start Cura

**Actual Results**
Cura doesn't start even if the same Cura installation has started on the same PC before. After setting "Windows 8" compatibility on the Cura.exe, Cura starts fine again.

**Expected results**
Cura should start without the need for the Windows 8 compatibility beeing set manually.

**Additional Information**
See https://community.ultimaker.com/topic/27056-cura-36-and-40-dont-start/ for more reports of the same issue.

Ghostkeeper, 2019-04-04:
Maybe do you have a log file of Cura starting unsuccessfully (without the compatibility mode set)?

reibuehl, 2019-04-04:
The files %AppData%\cura\stderr.log and %AppData%\cura\stdout.log are created when I try to start cura without the compatibility flag but they stay empty. The file %AppData%\cura\4.0\cura.log is not created. The process is visible in task manager but no UI components are shown - not even the splash screen.

Yves31820, 2019-04-04:
J ai le même soucis

Ghostkeeper, 2019-04-05:
Very weird! Then maybe running this via the command line would give a bit more output to debug what's going on. Here's some instructions, though you seem to know your salt but just in case:

1. Press Win+R.
2. Type `cmd` and press enter.
3. In the command window, type `C:\Program Files\Ultimaker Cura 4.0\Cura.exe` (or wherever you installed it).

I'd expect it to produce something in STDCOUT then... :thinking:

reibuehl, 2019-04-06:
Unfortunately no output at all (see attached screenshot). The command prompt comes back very fast. Process Explorer shows that the process is running but no output on the command line.

[image]

mqudsi, 2019-04-06:
Run eventvwr.msc and check `Windows Logs\Application` for any Cura-related entries.

fwiw I'm running Cura 3.6 and 4.0 under Windows 10 64-bit edition (17763) without any such issues.

reibuehl, 2019-04-06:
I checked the Application event log but no entries there either.

JamieARK, 2019-05-02:
I seem to have this issue on the latest Windows 10 1903 build as well, the program appears briefly in task manager but then closes.  However running in Windows 8 Compat Mode does not work in this instance.  Having checked those logs you mention I do however have some entry to do with a fatal python error, is this related?
[stderr.log](https://github.com/Ultimaker/Cura/files/3137111/stderr.log)

There is no mention of any errors in event viewer also.  Happy to experiment as could do with a fix, we're an IT Company so reasonably technically capable and have recently sourced a Ender 3 printer to do some 3D Printing work for some of our school clients.

Apologies if this is unrelated, but it's the only bug/article that appeared that was similar upon a search.

Thanks
Jamie.

Ghostkeeper, 2019-05-07:
We're also getting this bug report via the support channel of Ultimaker (who gets it via resellers who gets it from customers).

JamieARK, 2019-05-07:
If it helps, my issue was actually caused by the fact my laptop has multiple graphics chipsets.  My laptop has both Intel Graphics on board but also a dedicated Nvidia 1060 as well.  If I configured the app to always start using the Intel Chipset it opens fine, it only has an issue with the Nvidia chipset.

It just seems to be this shared setup that is the issue as my desktop PC only has an Nvidia Card and that opens fine.

Thanks
Jamie.

Ark ICT Solutions Ltd
'Striving to be the best ICT Supplier
in the East Midlands'
Jamie Kelly
2nd Line Support & Server Specialist
Ark ICT Solutions Ltd
Unit 6 Venture Court, Pinchbeck, Spalding, Lincolnshire. PE11 3BG
Email: Jamie.Kelly@ark.me.uk
Tel: 01775 720252
Mob:+44 07912287134
Web: http://www.ark.me.uk
Consider the environment. Do you really need to print this email?
Ark ICT Solutions Ltd is a company registered in England and Wales under number 5356119. The company's registered office is Unit 6 Venture Court, Pinchbeck, Spalding, Lincolnshire, PE11 3BG, United Kingdom.This message contains confidential information and is intended only for the intended recipients. If you are not an intended recipient you should not disseminate, distribute or copy this e-mail. Please notify the sender immediately by e-mail if you have received this e-mail by mistake and delete this e-mail from your system. E-mail transmission cannot be guaranteed to be secure or error-free as information could be intercepted, corrupted, lost, destroyed, arrive late or incomplete, or contain viruses. The sender therefore does not accept liability for any errors or omissions in the contents of this message, which arise as a result of e-mail transmission. If verification is required please request a hard-copy version.

TheAndi, 2019-05-08:
It also happens on my notebook with Intel Core i7-8550u and **just** Intel UHD graphics 620.

This bug happened to me in 3.5, 3.6 and 4.0 (and it also takes an eternity to open :/ )

Thanks in advance.

Rickrat, 2019-05-12:
> It also happens on my notebook with Intel Core i7-8550u and **just** Intel UHD graphics 620.
> 
> This bug happened to me in 3.5, 3.6 and 4.0 (and it also takes an eternity to open :/ )
> 
> Thanks in advance.

yes. me too on my microsoft surface 2017 model (windows 10)

Donner-Party, 2019-05-14:
> If it helps, my issue was actually caused by the fact my laptop has multiple graphics chipsets. My laptop has both Intel Graphics on board but also a dedicated Nvidia 1060 as well. If I configured [...]

I think I have this problem, too. How do you make it use Intel rather than Nvidia?

Donner-Party, 2019-05-18:
I disabled Intel UHD 620 in the control panel and Cura 4.0 now runs. #5699

reibuehl, 2019-05-18:
I have the problem on a Desktop PC with only one display adapter according to device manager. It is an Intel HD Graphics 630.

trunneml, 2019-05-31:
Same here. (MS Surface Pro 4; Intel HD Graphic with dual screen setup)

Adminius, 2019-07-21:
same here (i3-6100 with integrated Intel GPU), Windows 10 x64.
Tested with Cura 3.6 and 4.1

italocjs, 2019-07-28:
Any luck with this?  interestingly i also have intel UHD620 as graphic card (notebook), so this issue might be related to it

Donner-Party, 2019-07-28:
No luck for me. I still have to disable UHD 620 when I want to use Cura.

italocjs, 2019-07-28:
@Donner-Party  you have another graphic card right? i dont think thats a option for me, i made it work with win7 compatibility, but its kinda slow

Donner-Party, 2019-07-28:
Yes. My laptop has dual graphics processors.
[image]

johnseghers, 2019-08-23:
This thread on the Ultimaker forums seems to be confirming an issue with Intel GPUs. https://community.ultimaker.com/topic/29138-cura-421-does-not-start-and-makes-41-dead-as-well/?tab=comments#comment-245633

I have a SurfaceBook v2. This has an Nvidia GTX 1060 in the detachable keyboard and an Intel UHD 620 in the screen. When the screen is detached (Intel GPU only) Cura will not start. When connected to the keyboard it starts fine.

I did have an issue where the Nvidia adapter would stop responding. When this happened, again with only Intel GPU available, Cura would not start.

When Cura fails to start like this, there is no log information written.

ScruffR, 2019-09-10:
Yup, same here on a Surface Pro 2017

ScruffR, 2019-10-06:
This issue still exists in 4.3! 
Is this ever going to be addressed?

Ghostkeeper, 2019-10-07:
We're trying to fix this. Since 4.3 it also started reproducing on our test printers. We still have no idea what's causing it though. And it's a bit of a phantom bug, too, as is evident by you reporting that it works in 4.1 but not in 4.3 while the original issue was reported for 3.6 and 4.0.

Pseudothink, 2019-10-09:
This is happening with me on my fully updated Surface Book 2 (Core i5 with integrated Intel graphics only) with Windows 10 and Cura 4.1.  If I start Cura, the Cura process appears in task manager, but never opens a window.  If I start another Cura instance, it does the same thing.

oferfrid, 2019-10-10:
same here on my HP EliteBook 840 G5

Cesar-S, 2019-10-11:
Hi, install in mode compatibility 

[image]

reibuehl, 2019-10-11:
I tried exactly the same settings but even with the compatibility settings set, it doesn't start on my HP Elitebook 8470p with built in Intel graphics. It shows the splash screen with starting user interface, then opens a bigger, completely white window and then crashes.

Cesar-S, 2019-10-11:
In my case it is UHD 600 Intel N4000, I hope it continues to work.

Donner-Party, 2019-10-11:
With Intel UHD 620 enabled, the following programs won't run properly on my laptop:
- Autodesk Fusion 360
- Autodesk DWG Trueview 2019
- Ultimaker Cura 3.x & 4.x

This was not a problem before April 2019. I think a windows update caused my grief.

I think we need to complain to Intel or Microsoft rather than Autodesk or Ultimaker/Cura. I haven't found an Intel support forum that can suggest anything better than update drivers and reboot. When I try to install an older driver from HP's website, it is not recognized as a correct driver for my laptop.

smartavionics, 2019-10-12:
FYI see https://community.ultimaker.com/topic/29797-experimental-windows-cura-build-based-on-pyqt-513/ maybe it's worth trying?

Adminius, 2019-10-13:
@smartavionics 
thx! it works for me now (i3-6100 with 530 GPU)

smartavionics, 2019-10-13:
Hi @Adminius , thanks for the report. Please report again if you have any further problems related to this.

reibuehl, 2019-10-15:
Hi @smartavionics 
 
> FYI see https://community.ultimaker.com/topic/29797-experimental-windows-cura-build-based-on-pyqt-513/ maybe it's worth trying?

I tried that version but for me it doesn't fix the issue... I still get the splash screen with loading machines, loading interface, then a blank window that then crashes.

smartavionics, 2019-10-15:
Hi @reibuehl , thanks for the report, sorry it didn't help you.

nallath, 2019-10-16:
Could you also try to run cura but then via commandline? Some things are logged before the logging plugins are loaded (so they don't get written to the semi permanent logs).

reibuehl, 2019-10-16:
Hi @nallath , if I try that, the command prompt returns immediately. No logging there. The Cura.exe seems to just run the python stuff that then shows the splash screen and everything after. Should I create a short video of the behavior?

Cesar-S, 2019-10-16:
Hi, I had a similar problem with another slicer.  
You have to go to the system registry, delete the records and leave that window open.  
Open Cura, go to the registration window and update data, if there is an error it will appear there.

BobBuenzow, 2020-02-05:
Similar issue here.  Cura 4.4.1 installed fine on Win 8.1 64bit.  Upon execution, it hangs during interface load, then shuts down.  Log file attached.  Thanks in advance for any advise.
[cura.log](https://github.com/Ultimaker/Cura/files/4162127/cura.log)

Cesar-S, 2020-02-05:
Try running curaengine.exe and then cura.exe ... it happened to me in the latest version and I only did it once, then I charge well.

BobBuenzow, 2020-02-05:
Cesar,

 

Thank you very much for the quick reply and advise.  Unfortunately, it did not fix my problem:

 

 

Do you know anything else I might try?

 

Best Regards,

 

Bob

Cesar-S, 2020-02-06:
have you tried this? it was the best result for me

[image]

BobBuenzow, 2020-02-06:
Hi Cesar,

 

I was using those settings but they did not get the program to start.

 

I found my solution.  I had to turn on the high performance settings on my NVIDEA graphics processor.  Once I did that, it started up properly.

 

Maybe this info will be helpful to others.  Thanks for your help.

 

Bob

 

 <https://user-images.githubusercontent.com/4626896/73895128-47ae8700-485d-11ea-8eef-39dd2fc5b172.png> 66627552-70d24a00-ebd2-11e9-960c-6edb2e548ec8 1

reibuehl, 2020-02-06:
I tried setting my INTEL HD graphics to high performance but Cura still doesn't start. All versions from 3.x to 4.4.1. Cura 15.04 runs fine without issues.

italocjs, 2020-02-06:
For me clicking like 20 times on the program eventually opens it, tho honestly i gave up on cura (at least on my personal laptop).

BobBuenzow, 2020-02-06:
Probably a compatibility issue with the INTEL HD driver.  That failed on my LT as well.  I also have an INVIDEA GeForce card installed.  Enabling 3D acceleration on the GeForce fixed my issue.

 

Glad you can run with 15.4

Ghostkeeper, 2020-02-11:
As far as we know, this issue is fixed in the 4.5 version. We'll release a beta today, so feel free to give that a try. It's a complicated issue since a lot of people are reporting different things under the same denominator.

reibuehl, 2020-02-11:
I just installed 4.5 beta and it still fails to start on my HP Laptop. Should I try any special settings or is there anything I can do to help you fix this issue?

BobBuenzow, 2020-02-11:
Hi Reiner,

 

I’m not affiliated with Cura; just another user like you.  I also happened to have the same issue on my HP Envy LT (for any version up to 4.4.1; haven’t loaded 4.5 beta as yet).  What worked for me was to go into the NVidea control panel; Manage 3D Settings and select High-performance NVIDIA processor as the preferred graphics processor on the Global Setting tab.  Auto-select will default to integrated graphics and GUI load will fail.  Apparently the imbedded graphics processor is not up to the task.

 

Hope this helps!

 

Good luck,

 

Bob

reibuehl, 2020-02-11:
@BobBuenzow I have INtel HD graphics, no NVidia. I tried what seemed to be the equivalent in the Intel driver of what you suggested already earlier. My last comment was meant for @Ghostkeeper who seems to be one of the developers. I tried 4.5beta and was wondering if I can create some logs that help him and the other developers to identify the root cause of why it does crash with the Intel HD graphics on the HP Elitebook.

Ghostkeeper, 2020-02-13:
@riebuehl there are two things I'd like to ask you to try:
* Update your Intel graphics drivers. I think we've probably asked you this before, but a few weeks ago Intel came with an update to their graphics drivers that specifically fixed one of our issues that had this symptom.
* If Cura still doesn't start after that, kill the process from the task manager and please find the file `C:\Users\<riebuehl>\AppData\Roaming\cura\4.5\cura.cfg` and open that with a text editor. Under the header `[view]`, add the following line:

      opengl_version_detect = force_modern

Save that file and try starting Cura again. Does it start then?

reibuehl, 2020-02-13:
@Ghostkeeper I did already have the latest drivers from the laptop vendor but did now manually install the newer generic graphics drivers from Intel. Version 10.18.10.5100 was the latest Intel had for the Intel Graphics HD 4000. With this driver version, Cura 4.4.1 and 4.5beta both start.
I will verify if this also solves the issue on my other system that has a different configuration on Saturday and close the issue then if successful.

marcus-bigrep, 2020-02-14:
@Ghostkeeper Cura 4.5 states something was fixed with Intel GFX drivers. 
Is this about this issue? And what is the related commit?

reibuehl, 2020-02-14:
Installing the latest (as of February 2020) graphics card drivers from Intel fixed the issue for me on both systems where I had the issue. One had the Intel Graphics HD 4000 and the other the Intel Graphics HD 630.

closed 2020-02-14 by reibuehl

Ghostkeeper, 2020-02-17:
> Cura 4.5 states something was fixed with Intel GFX drivers.
Is this about this issue? And what is the related commit?

We've fixed multiple things in 4.5 that are related to Cura not starting. One of them is the workaround I posted above with `force_modern`. Related commits are here: https://github.com/Ultimaker/Uranium/pull/565

Ghostkeeper, 2020-07-20:
Did you try Windows 8 Compatibility? Or the workaround I mentioned above?

Mutos, 2020-07-22:
Thanks  Ghostkeeper for your answer !

I've left the issue for now, got another machine on which Cura works OK and no time to test. This machine is usually on Linux and rarely booted on Windows ^-^ But this week-end I'll probably have enough time to test both workarounds. I don't know if there is a W8 compat setting on W7, but I'll look... but the `force_modern` workaround, which seems best bet...