Ultimaker/Cura #5571: Per object retractions

guestisp, 2019-04-04:
Would be possible to add a per object retraction feature ?
Would be pretty usefull when calibrating a printer, just have to add multiple object with different retraction settings, like with Simplify3D

A single print is much faster than tons of different prints with different settings

Ghostkeeper, 2019-04-05:
What needs to happen when it makes a travel move from one object that disabled retractions to an object that enabled it? Or vice versa?

guestisp, 2019-04-05:
> What needs to happen when it makes a travel move from one object that disabled retractions to an object that enabled it? Or vice versa?

I think would be indifferent, as retraction test are per object and not between object. Maybe keeping the last used retraction would be good. But let's see what Simplify3D does....

guestisp, 2019-04-05:
If you stack object vertically, like a retraction tower, you don't have to change the retraction between object as there is no retract, vertically

GregValiant, 2026-09-07:
I'm cleaning up these old reports.  I'll go ahead and close this.  This effect can be obtained by setting the "Per Model" settings of any model, or using "Tweak at Z" or the older "Change At Z".

closed 2026-09-07 by GregValiant