Ultimaker/Cura #5572: Nozzle undefined when saving project

Nekksys, 2019-04-04:
Whenever I save a project in Cura, the nozzle always shows as "undefined" on the summary page.

**Application Version**

Cura 3.6 and 4.0

**Platform**

Linux (Ubuntu)

**Printer**

Creality Ender 3 

**Steps to Reproduce**

Create printer. Check settings. Create material. Check material settings. Load STL. Save project. Nozzle is undefined in summary.

**Actual Results**

Nozzle is undefined in sunnary

**Expected results**

Nozzle should be defined as per settings in machine; in my case, 0.4mm.

**Additional Information**

See attached screen caps.
[image]
Creality Ender 3 settings, printer with adjusted bed size defaults.

[image]
Creality Ender 3 settings, extruder; note nozzle size is set to 0.4mm.

[image]
Save project, summary window; note nozzle size is undefined.

nallath, 2019-04-04:
You're right, but only partially. Your machine doesn't have a selectable nozzle type (in the way that some printers have it, such as the um2 or um3), so it should not be shown at all.

This has been fixed for the next release (4.1)

closed 2019-04-04 by nallath

Nekksys, 2019-04-04:
So basically, it was an oversight on the part of the Cura programmers?

nallath, 2019-04-05:
No, it is using it. We also have a thing called "variant profiles", which actually changes what profiles are used based on what nozzle type you have selected. Those profiles are pre-defined.

Most third party profiles don't have that well-defined profiles and thus don't use all the cool features that are out there. This is also the case for the ender profiles.

Ghostkeeper, 2019-04-05:
Your printer defines a nozzle size, which is saved with the project. You don't have different built-in profiles for different nozzles, so you can't select that, but you might have changed the nozzle size yourself and that would just get saved right along.

Many people identify a "nozzle" only by the size of the opening in the bottom, but Cura sees a "nozzle" as a profile which could be any collection of settings. Nozzle size is not a special setting in any way. It gets saved right along with the profiles. We have no reason to display just the nozzle size there. For instance, the Ultimaker S5 has a nozzle profile for AA0.4 and BB0.4 which have the same nozzle size but have different other properties.

Nekksys, 2019-04-08:
OK. There is, however, a very sound reason to display a single head printer's nozzle size in the summary: so it doesn't look like an error. To anyone with programming experience, "undefined" is an error to be avoided at all costs. Glad you got that sorted for v4.1!

commit 2dd4a305 (2019-04-04): Prevent machines without an extruder from showing "undefined" in the workspace dialog

This fixes #5572