Ultimaker/Cura #5573: Printing concentric walls and infill, in as few passes as possible

yet-another-average-joe, 2019-04-04:
This is more a feature request than a bug report !

And sorry for my bad english, I miss some vocabulary. Hope you will understand what I mean !

**Application Version**

Cura 4.0,  and probably previous versions

**Platform**
Windows 10

**Printer**
Tevo Tornado + DIY 1 stepper + belt driven dual Z + E3D V6 from E3D Online. Everything is perfectly callibrated using real metalworking machine shop metrology instruments. The printer is for hobby use.

**Steps to Reproduce**

The context is the design of IP66 enclosures for electronics, electrical components, machine tools command boxes (the reason why I started with 3D printing). The goal is to create models that will print fast, with a large nozzle (at this time, tested 0.4, 0.6 and 1mm, PLA and PETG). Think of a classic junction box or machine commands enclosures, with lips, gasket grooves, and 3mm thick sides, infill being 100%, top lips being 1mm thick. All filets, inner and outer are all concentric. Everything is 1mm based, the nozzle diameter.

In the archive, I put a small step model, its stl export, and a 3mf file with some parameters tweaked in order to get something printable.

The box is designed with dimensions that are a multiple of the nozzle diameter

**Actual Results**

Cura does not print the top inner lip (1mm thick), even with thin walls activated. But it does work if inset outer wall is set to -0.01mm (negative)

Cura makes the nozzle do twice as printing travels as needed, goes back and forth for 1mm thick lips, and even generates holes (see layers 23 to 27)

I printed about 20 models, changing parameters including some dimensions in the model (slightly thicker sides and lips), and got some good cosmetic (but not functionnal) prints. At this time, the only way to go is to print with small nozzles. 

With default parameters, the inner lip is not printed, and the outer one is printed... how to say ? Weird ! 

A view of the CAD design :

[image]

The results :

[image]

[image]

The sources and the 3mf file :

[test 02.zip](https://github.com/Ultimaker/Cura/files/3044729/test.02.zip)

"test 02 weirdest.3mf" shows the really weird behaviour if outer wall inset == 0.00

**Expected results**

Cura should print concentric walls and infills in one pass only. It should also print the lips without going back and forth, and without holes..

The result is actually not that bad, but will never be 100% IP66. The only way to print such parts perfectly is to use a smaller nozzle. A large nozzle is interesting for such parts because they are generally large themselves (like 250x200mmx150), and have to be ~3mm thick,  and 100% infill. Small nozzles are a huge waste of time. But they do the job (and they do it great)... 1.0mm with 0.7 layers (tested) could give great results.

Results are better with PETG (yellow) than PLA (blue). Probably because of its higher viscosity. I doubpt that linear advance would help a lot, as holes already are in Cura preview...

**Additional Information**

This is more a feature request : I imagine something like a "thick vase mode", but working layer by layer, printing concentric layers in the XY plane, and building concentric infill with minimal movements. Maybe a great function would be a 100% concentric infill between the walls, going up to the topmost layer ?

smartavionics, 2019-04-04:
Your .3mf files only contain the model, not the profile so I have none of your settings. Do File -> Save to save the whole project (model and settings).

Have you tried using the Minimum Wall Flow setting? It works in conjunction with the wall overlap compensation. With the overlap compensation enabled, setting a min wall flow value of, say, 50 will stop the 2nd pass over the walls that should only be 1 line width thick.

smartavionics, 2019-04-04:
Are you using coasting? Not sure that works well.

yet-another-average-joe, 2019-04-04:
Sorry ! I realize I did an export, not a save !

The files :

[new.zip](https://github.com/Ultimaker/Cura/files/3044825/new.zip)

I don't use coasting.

I will give it a try within a couple of hours (GMT +1). I understand say 66% of the 300 parameters at this time ! There was a great help plugin, but it's gone with 4.0...

smartavionics, 2019-04-04:
Those weirdy gaps in your outer wall are because the model's wall is slightly thinner than the line width at that point. If you either make the wall a tiny bit wider or slightly reduce the wall line width the gaps will go away (well, they did for me!)

smartavionics, 2019-04-04:
Don't use z-seam alignment set to sharpest corner, it very rarely does what you want. Instead, use the user specified coordinates.

yet-another-average-joe, 2019-04-04:
Thanks !

> the model's wall is slightly thinner than the line width at that point

Got it ! There were 2 problems in the STL. First I edited it so much times that I forgot to canceI the last mod before I post it (inner single wall). And most of all, I re-exported the STL with maximum accuracy, and the holes disapeared. I did'nt realize the export was so coarse by default. The top inner single wall is now there, without the negative inset trick, and without the nozzle jumping all around the place. I believe Cura was jumping from messy triangle to messy triangle. It was the STL. Now 2 microns and 1° resolution. Still to find the sweet spot (the file size exploded).

> Minimum Wall Flow

googling these words, I found a discussion on the Ultimaker forum (still to be read and fully understood, but I learned that the "lips" are called "single walls"). With outer before inner, it solves the problem for the groove sides (single walls) and the box corners. Tested with PLA and PETG, with 0.7 and 0.5 layers. Still not perfect beccause of the 100% infill :  the amount of material has to be slightly decreased, no big deal.

> Seams

I set them to random. It had no effect. Had to restart Cura. Weird...
(printing 99% structural parts, I prefer having them aligned, easier to post precess if needed)

But Cura still asks for the nozzle to pass 2 times on body corners. The rounded corners are perfectly concentric, and the infill is exactly 1mm wide everywhere ; I'm aware that being rounded they need more material on the outside than in the inside of the curve. I had some success playing with the radii, but it would make the printed item unusable.

Having an option for deactivation of this compensation could still be interesting, as well as a vertical preset concentric 100% infill  : a rounded rectangular box being more rigid at its corners, some underfilling is not a problem, and it would significantly reduce printing time and travels. 

Or the nozzle could do the outer path on one layer, and the inner one the next.

Thanks again !

Ghostkeeper, 2019-04-05:
> Having an option for deactivation of this compensation could still be interesting

Look for the setting "Compensate Wall Overlaps" and disable it.

> as well as a vertical preset concentric 100% infill : a rounded rectangular box being more rigid at its corners, some underfilling is not a problem, and it would significantly reduce printing time and travels.

It is an option right now (you can change the skin and infill pattern or just set the wall thickness very high) but it's not the default because in a lot of cases, like your cube, it produces a big blob in the middle where all of the concentric contours converge. The problem is that the nozzle stays in the middle for so long that the material there starts to degrade and also any extra ooze has no way to escape. The blob then often knocks over the print, or if you're lucky just ruins the visual quality.

> Or the nozzle could do the outer path on one layer, and the inner one the next.

If not all layers have the same shape, this is infeasible.

yet-another-average-joe, 2019-04-05:
Compensate Wall Overlaps cannot be set to false, as it kills the one pass single walls.

Applying all suggestions, I finally got the PERFECT settings.

(nozzle = 1.0mm)
Compensate Wall Overlaps = false
Outer Before Inner Walls = true
Wall Thickness = box side thickness i.e. 3mm -> wall replaces infill, therefore infill is concentric, 100% and *printed in one pass* (it is weird, I had already been testing this, and completely missed the result). 3mm only because cross pattern for the 1st layers is sturdier).
Minimum Wall Flow = 50%
Z Seam Alignment was set to Random, but it had no effect. Seams are aligned, even after restarting Cura... 

The results are excellent. I printed 2 test models, with 0.5mm layers. 30x30x13.8mm. The first one (blue) is PLA, with 95% flow (I decreased a little too much). But I consider it is perfect or near to (inside blobs or zits are not a problem). Maybe it will be even better when I receive the Volcano hotend.

[image]

The second (black) is PETG. First reel from a brand I don't know. I used black one because black does forgive nothing. It is just 99.99% perfect. Just some tiny imperfections, barely visible by eye, exagerated by the DSLR and the 2 photo lightings tuned in order to show them.

[image]

And it prints at light speed, with no unwanted paths.

Layer and walls adhesion are 100% perfect. I crushed the 2 models in a large vice. The PETG exploded (safety glasses !). Layers and walls cannot be seen, it looks like it is injection molded. I couldn't break the PLA one.

The settings (for PETG) :
(internal profile name is irrelevant)
[test 02 final.zip](https://github.com/Ultimaker/Cura/files/3047990/test.02.final.zip)

So, You are right, there's no need for a special mode...

But I still have one suggestion. Could the profiles contain an open text field for notes and observations in a future realease ? It would be really handy. And even for printer profiles (workaround for printer : notes can be written as comments in the GCode fields)

I will try again to create a first post on the forum...

Thank you all !

Ghostkeeper, 2019-04-11:
The notes for profiles is already requested here: https://github.com/Ultimaker/Cura/issues/3424

> Compensate Wall Overlaps = false
> ...
> Minimum Wall Flow = 50%

You shouldn't be able to set a Minimum Wall Flow if Compensate Wall Overlaps is disabled.

yet-another-average-joe, 2019-04-11:
Do you mean that Minimum Wall Flow should be disabled if Compensate Wall Overlapse is set to false ?

I think I mistaken when I wrote "Compensate Wall Overlapse = false".

I downloaded the archive I previously uploaded, set the nozzle to 1.0mm : Compensate Wall Overlapse was set to true. Otherwise the "infill" inner wall and the top single walls require 2 passes. I edited my previous post !

Ghostkeeper, 2019-04-12:
> Do you mean that Minimum Wall Flow should be disabled if Compensate Wall Overlapse is set to false ?

Yeah the setting should disappear from your interface.

If you feel like changing the settings are a sufficient solution, perhaps we can consider this issue fixed.

yet-another-average-joe, 2019-04-12:
It does not...

[image]

Should I open another issue ?

I close this thread, as the problem is definitely solved 👍

closed 2019-04-12 by yet-another-average-joe

Ghostkeeper, 2019-04-12:
Yeah. I'm not able to reproduce that though. In the setting definition it says that the setting is visible if either Compensate Outer Wall Overlaps is enabled or Compensate Inner Wall Overlaps is enabled. Maybe some weirdness with the display of the checkmarks. We've had that before.

philipphirte, 2022-11-10:
@yet-another-average-joe 
Where is the compensate wall overlaps setting & minimum wall flow setting?
I'm in V.5.1.1 and its not showing for me.

yet-another-average-joe, 2022-11-12:
@philipphirte 
It was 3+ years ago, Cura 4 !

Since then, Arachne came out and was a game changer for such prints !