Ultimaker/Cura #5574: [4.0.0] Monitor Page Broken on MacOS 10.14.4

maalencar, 2019-04-04:
This is a weird one since its was working right after I installed 4.0.0, but for the last few days when I click on the monitor tab it shows a little photo camera icon instead of the usual video camera on Cura Connect and the iOS app. If I click on the icon nothing happens. Also, the links "Go to Cura connect" and "View Print History" don't do anything.

Application Version
4.0.0

Platform
Mac OSX 10.14.4

Printer
US5

Steps to Reproduce
Try to print anything and go to the monitor tab.

Actual Results
Dead links and no ability to monitor print.

Expected results
Page to work properly

ianpaschal, 2019-04-05:
Can you upload some images of what you're seeing? Also, do you mind telling me what firmware is on you printer?

Ghostkeeper, 2019-04-05:
I have a similar issue at home, which is caused by my printer's camera hardware not functioning. It could be an issue on the printer's side.

Those links should open a browser page. Maybe the browser was somewhere on the background?

ianpaschal, 2019-04-05:
> It could be an issue on the printer's side.

This is my thinking as well. If it changed icons on Cura Connect and iOS (which is a webview), and it _was_ working fine on Cura 4.0.0, then I suspect there's an issue with the newest firmware (specifically the version of Connect which shipped with it).

closed 2020-06-20 by no-response[bot]