Ultimaker/Cura #5579: Cura 4.0.0 - Top surface low quality

amagr0, 2019-04-05:
**Application Version** - CURA 4.0.0
**Platform** - Win 10 Pro, 64 bit
**Display Driver** - NVIDIA GeForce 840M

Hello,
Is there any way to avoid this:

[image]

[image]

[image]

[image]

[image]

The text "WHY?" will only be printed on 4th layer, but in the previous layer Cura start to do something like a tracing of the text.
This behavior increases a lot the printing time, and worst than that, decrease a lot the quality of the top surface of the 3th layer.

I think this was not always how Cura worked , but somehow this changed.
Is there any obvious reason to why this was changed?

Is amazing how complex the path is becoming to solve something that in the past was so simple.

Thanks

smartavionics, 2019-04-05:
It's because you have top skin layers (or whatever it is called) non-zero.

amagr0, 2019-04-05:
But if put zero the benefits of this setting will be lost, like for example, the control of the print speed of the top surface.

smartavionics, 2019-04-05:
That's right. If the top surface has features like the text in your example, the top skin layers feature doesn't really work well.

smartavionics, 2019-04-05:
It needs to be smarter to know that the skin layer(s) below the letters should also be treated as top skin layer rather than just underlying skin layer.

Ghostkeeper, 2019-04-11:
Well, but then it's going to print the rest of the skin at a lower speed as well. You might as well just set the top/bottom print speed lower then.

Because these top skin lines can have different line width, speeds, etc. they will get planned separately and thus not necessarily printed in order. After all, it's completely impossible to print the lines continuously lengthwise if they change to a different line width. In your example, the "why" text on layer 3 is printed with the normal skin settings but the rest of the layer is printed with top skin.

I don't really see a way to prevent this. Trying to detect what the user intended typically causes more damage than it heals because it focusses on specific geometries and makes the whole system more complex.

closed 2026-09-07 by GregValiant