Ultimaker/Cura #5580: [4.0] Wrong material assignment for dual extruder machine

yyh1002, 2019-04-05:
**Application Version**
4.0

**Platform**
Windows 10

**Printer**
Ultimaker 2+ DXU mod
gMax 1.5 Plus Dual Extruder

**Steps to Reproduce**
Load a model, set it to extruder 2 (blue). 
Change adhesion and support extruder to extruder 2 (blue).
Slice the model. 
Basically attempting to do a pure extruder 2 print without heating up extruder 1.
[image]

**Actual Results**
The tool path shows color of extruder 1 (yellow). 
[image]

**Expected results**
The tool path shows color of extruder 2 (blue). 

**Additional Information**
I tested with both the Mark2/DXU for UM2 custom machine and the included gMax 1.5 plus dual extruder, same problem occurs. 
I've been doing this in 3.6 with no problem. Only 4.0 has this issue. 

Thanks

Block137, 2019-04-05:
#5550
Discovered just few days ago

closed 2019-04-11 by Ghostkeeper

nallath, 2019-05-03:
Fixed for 4.1