Ultimaker/Cura #5581: 4.0 release errors

Thisismydigitalself, 2019-04-06:
**Application Version**
4.0 R

**Platform**
Win10/64

**Printer**
CFFFP

**Steps to Reproduce**
Just launch Cura 4.0 R (fresh install - side by side to 3.6)

**Actual Results**
[image]

**Expected results**
feel free to guess.. :)

**Additional Information**
never had this issue with 3.6 which i am going back to. i did had these errors with previous versions. i guess old isseus die hard.

nallath, 2019-04-08:
Could you share the logs / your configuration?

Thisismydigitalself, 2019-04-09:
[cura.log](https://github.com/Ultimaker/Cura/files/3060694/cura.log)

rburema, 2019-04-10:
A settings-upgrade wasn't necesary to go from 3.5(.1) to 3.6. I suspect that's why you didn't see it then, since upgrading settings seems to be what's going wrong now (and likely has been in the past).

Another clue is all the 'failed to get file version' for the materials profiles I see: Is there a version tag in those custom materials files?

That still leaves the actual problem of the extruders: The only thing in the log is that they're added to the machine, but that'll just be ignored. Extruders don't reference the materials directly (as far as I know) so its not likely to be some sort of cascade.

In short: Cura.cfg isn't the file we'll need to diagnose, it's those custom_extruder_* configs, a bit deeper in the %appdata% folder :-)

P.S. From browsing through the log: Your OctoPrint plugin is outdated. (4.1 will have a message if an update is avialable for installed plugins, but 4.0 is lacking this)

Thisismydigitalself, 2019-04-10:
Thank you.
once i get back home i will follow your suggestions and report back. just for the record i always try to keep new versions as clean install side by side to previous versions so i am not sure why there is this error about bad settings-upgrade reported in cura log file.

rburema, 2019-04-10:
Hi @Thisismydigitalself,

1. If you have an older version folder next to the new one, it'll try to upgrade all your old settings to the new one.

2. What do you mean with 'zigzag issues'? We fixed a few things in 4.0, but in order to do that, right now, we're a bit too strict on when _not_ to merge 'arched' or 'zigzagged' lines into a new single line.

Thisismydigitalself, 2019-04-10:
@rburema 
1. How do i upgrade my old settings without touching / altering my 3.x data?
2. like with 3.6, when you set zigzag for top/bottom layers the lines do not overlap and leaves gaps. lines works as expected. pleane note that 3.4.1 top/bottom zigzag works as expected and good.

here's a sample of my custom extruder file.
i have so many of these. i wonder why.

[general]
version = 4
name = custom_extruder_4 #5_settings #4
definition = custom_extruder_4

[metadata]
type = definition_changes
setting_version = 6

[values]

rburema, 2019-04-10:
Thanks for the config info! :-)

1. I'm not sure what you want to do here: Do you have an even older version that you do want to keep the settings of in 4.0, but skip all the settings that you've made for 3.6? Or just have a 'clean' install of 4.0 next to the existing 3.6 one?

2. Could you please make a different issue for the 2nd point? It'll be a bit hard to refer to one specific issue otherwise. ("No I mean the _other_ thing the user reported ten replies down the issue." is not always a fun game to play....)

Thisismydigitalself, 2019-04-10:
This issues is not about the zigzag i mentioned. it was a side note.
This is about these errors that i get each time i launch 4.0. What sould one practice to get rid of these?

Thank you.

rburema, 2019-04-11:
I understand you want a clean install of 4.0 besides 3.x: The only way you can do that is (before install of the new version) by temporarily removing the (older) settings files, then pasteing them back after the install has made the new 4.0 folder. There is now way to do this without 'touching' those files (but they don't need to be altered).

To get rid of the errors, the simplest way is to either;
 - reinstall 4.0 (with the above procedure around it)
 - do the above procedure, but instead of reinstalling, remove the 4.0 settings folder, then restart
 - try to remove only the offending settings files (least likely to work, but maybe worth a shot if you already have things in 4.0 you want to save)

We'll eventually plan to have an upgrade system that handles error conditions slightly more gracefully (so that it doesn't become an 'all or norhing') but it will take us a while to get to that point.

Thisismydigitalself, 2019-04-11:
Thank you. i followed all your suggestions to no avail.
Cura complains about these profiles which... actually works, but you never know what goes on under the hood - these might be curropted and effects the generated gcode - I'm in the dark here.
Anyways, no time to mess with this. I'll continue with 3.6 and/or 3.4.1 just for top/bottom zigzag patterns.

rburema, 2019-04-11:
On rereading it, it may have been my instructions that where not completely clear:
So you (temporarily) removed _all_ older folders from the %appdata% folder (or wherever that is on your system) and _then_ did a clean 4.0 reinstal, then started 4,0 _before_ you put them back? Becasue there is then no way that the installation procedure (and thus the new 4.0) can know of those profiles.

Thisismydigitalself, 2019-04-11:
Thanks, Just tried your latest procedure. still, no good. 
got completly different outcome and nothing close to my desired settings.
[image]

Sorry mate. too many times i dealt with new version's issues.
Will go back to 3.6 which was very good to me.

Ghostkeeper, 2019-04-12:
If you'd have properly deleted those files, Cura wouldn't be able to show any profile names. It doesn't think of those names by itself; they were written by you. Thus the only explanation is that you didn't properly delete those files.

Thisismydigitalself, 2019-04-12:
i was told to copy my data files back after reinstalling and running 4.0 clean the first time. i need all my 3.6 profiles in 4.0.

Ghostkeeper, 2019-04-12:
Ah, we understood earlier that you wanted to have a clean install of 4.0 (while still being able to use 3.6).

In that case, you're probably better off trying to delete only the files mentioned in the pop-up. If you restart, you'll find that it mentions more files to delete afterwards. But basically the situation that you're in has your configuration corrupt in some way.

I don't know how it got corrupt as the log doesn't say that. Unless you happen to know how it got in that state, but that could be a long history...

Thisismydigitalself, 2019-04-12:
Sorry guys, I believe I've tried it all. deleting / moving files, clearing cache.. you name it.
Nothing fixes and eliminate these errors. i have no idea what broke as up until a week ago i did not have these errors. I have no idea what else to do and I was hoping you guys based on the log file and error messages would tell me where to look. I have to leave 4.0 behind and continue using 3.6 and/or 3.4.1.

MoonCactus, 2019-04-17:
Somehow similar issue here. I eventually completely deleted my profiles for 4.0.0 on linux. Everything worked fine for a while... but I started to get some new "Your configuration seems corrupt"...
Moreover, the more I use cura, the longer the list of corruptions. Give the names it looks like it is duplicating profiles abusively and in a broken fashion.

Right now I have these errors when I open Cura:
  * ultimaker3_extruder_right_#2_empty
  * ultimaker3_extruder_right_#2_empty #2
  * tridimakeum3_empty #3
  * tridimakeum3_empty

It looks like Cura shoots in its own foot there (maybe on "abrupt" close of the app? I was not able to reproduce it in consistent way).

Now I could re-set my profiles again, as I already lost them in the upgrade process after trying to have them upgraded automatically in vain (for the same reason). But I guess the issue will come back.
It was already cumbersome to copy my fine-tuned settings manually when I realized a fresh start might fix the upgrade issue, but if Cura 4.0.0 corrupts the new settings as well by itself, it really gets annoying!

Ghostkeeper, 2019-04-17:
We've seen some hard-to-reproduce issues when connecting Cura to the printer that caused this symptom the other day.

However MoonCactus' case with the `_empty` profiles is really weird.

Thisismydigitalself, 2019-04-18:
For the record, I never connected 4.0 via usb/eth/octoprint to my printers. 
I only generated gcode and saved to file.

Thisismydigitalself, 2019-07-31:
I have same issue with 4.2B.
It worked ok for a day, then and after opening 3.6 projects i get a long list of corrupted configurations errors. can't remember ever seeing these errors with 3.6. 

[image]

Thisismydigitalself, 2020-10-20:
https://github.com/Ultimaker/Cura/issues/5706#issuecomment-711331051

closed 2020-10-21 by rburema