Ultimaker/Cura #5582: Incorrect slicing of threaded object

rftghost, 2019-04-06:
**Application Version**

4.0.0

**Platform**

Win7 x64

**Printer**

Anycubic Kossell Linear Plus

**Steps to Reproduce**

Open the stl file, slice.

**Actual Results**

Examine the preview for the gcode, print it.

**Expected results**

The generated gcode should reflect the initial geometry. The printed object should look like
a bottle cap, and not like a hexagonal chunk (no thread).

**Additional Information**

Tried to change the object in blender, checked geometry, all the good stuff. Tried all sort of
settings, no avail. The only way to "clean" the object was to tilt it, and Cura sliced it like it had water inside (water level filling).
[untitled.zip](https://github.com/Ultimaker/Cura/files/3050721/untitled.zip)
[D_untitled.zip](https://github.com/Ultimaker/Cura/files/3050723/D_untitled.zip)

smartavionics, 2019-04-06:
Hello @rftghost , setting the slicing tolerance to exclusive appears to fix the problem.

[image]

smartavionics, 2019-04-06:
BTW, I printed it and it came out really nice!

rftghost, 2019-04-07:
I can confirm the fix, everything sliced up nicely!
Thanks for the quick feedback!

Ghostkeeper, 2019-04-11:
That's a workaround, but not really a fix. I wonder what causes this because there is no geometry there that closes the mesh up. MeshLab doesn't even load the entire model for me. CloudCompare does, but indicates that the normals in the centre are the other way around. Maybe that's the problem? Though CuraEngine shouldn't mind that.

LipuFei, 2019-04-15:
For dev, JIRA ticket: CURA-6461

rburema, 2019-04-18:
Can also be fixed with the Meshtools plugin, then 'Extensions' > 'Mesh Tools' > 'Fix model normals'.

diegopradogesto, 2019-05-03:
This looks like a model dependent issue. The model has a lot of flipped normals. Actually the trick that @rburema comments is valid for this case.

closed 2019-05-03 by diegopradogesto