Ultimaker/Cura #5585: Have to wait for bed heating to finish before starting nozzle heating.

stefann82, 2019-04-06:
**Application Version**
4.0 - but likely every version previously as well.

**Platform**
Windows 7

**Printer**
All

**Steps to Reproduce**
1) Create a GCODE file - anything. 
2) Write the GCODE to SD card and run GCODE on printer.
3) Bed temperature is set (M190 SXX) ... there is a pause in code execution to wait for the bed to finish heating up (paused with M190 SXX).
4) After bed heat OK, Nozzle temp is set M104 SXXX... followed by a wait M109 S215

**Actual Results**
Bed heats up... I have to wait for bed heating to finish - on some machines (read: 12V heating beds) that means I have to grab a coffee, go for a walk, read a book, etc.... then afterwards nozzle starts to heat up. 

**Expected results**
Every single time I print with a cold machine I have to wait on the bed to heat before the nozzle heats up. I often manually go into my firmware control and set the nozzle temp (so it can preheat while the 'wait for bed heat' is active) but its annoying to have to do this every single print on every machine I print on.

What would it take to change the default behaviour to heat nozzle and bed at the same time... then wait for the heats to be met before printing:
1) Heat bed - M140
2) Heat nozzle - M104
3) Wait on bed temp - M190 
4) Wait on nozzle temp - M109

If you don't want to change the default behaviour, could we at least give an option to: "heat nozzle and bed at same time"

**Additional Information**
NA

fieldOfView, 2019-04-07:
You can add your own heating sequence to the start gcode in any order you want. If Cura does not see a heating sequence, it will add its default preheating sequence before the start gcode.

When you add your own preheating sequence, make sure you use the proper replacement patterns, like so:

```
M140 S{material_bed_temperature_layer_0}
M104 S{material_standby_temperature}
M190 S{material_bed_temperature_layer_0}
M109 S{material_print_temperature_layer_0}
```
Note that I sneakily added the standby temperature, so the material does not ooze and degrade while it sits in the nozzle waiting for the bed to heat up.

pcollingwood, 2019-04-07:
It makes perfect sense to do this. The bed takes much longer to heat. It would cause undue heating, drips and heat stressing of the filament if the nozzle was heated at the same time. After heating the bed for 10 min it takes about 2 min for my nozzle to heat.

stefann82, 2019-04-07:
Can we add an option to change the default heating sequence? 

We already have so many other options available to do many things that don't make 'perfect sense'.

Also I thought I was going crazy, but I did a google search of '3d printing heat stress filament' to see if this is really some phenomenon, and apparently it isn't. So maybe you can educate me on this. Also explain "undue heating" as opposed to "due heating"? 

Regarding the oozing. What happens if you're starting out with a filament change? In this case, you already just preheated your nozzle to remove the previous filament, and now you're pushing the new filament thru. Generally you want to load the nozzle up and see it oozing a bit, which is a good thing, because it validates that you loaded the filament correctly. Do you actually turn OFF your nozzle heater at this point, to stop the ooze, and then run the gcode or do you leave your nozzle running like most others?

pcollingwood, 2019-04-08:
Try to limit the sarcasm. This is an area where people try to HELP eachother.

I'll deal with your points in order:

Yes, there are many options. There are reasons for them. Not all are obvious, if you want this as an option and there is enough support for it I'm sure it will be implemented. If not, it won't.

Yes, there is such a thing as heat stressing the raw material. Just because you couldn't find it doesn't mean it's not a thing. I have this from my suppliers of ABS pellets because I make my own filament. Why do you think that when recycling dud prints, support material and roll ends all filament making machine manufacturers reccomend that you use at least 50% new material - Heat stress. Undue heating, incase you really don't understand, is heating without a due cause. Hopefully that explains the difference between due and undue.

Oozing. Having your hot end at full temp while the bed heats will cause the filament to slowly ooze out, assisted by gravity. This is not the same as priming the hot end, in fact it causes the opposite. While you may think this is a good thing what actually happens is that you end up with an empty hot end, which is no longer primed. The little filament that is left lining the bore and nozzle will tend to discolour due to - you guessed it - heat stress. 
How and when you personally heat and load your filament is up to you, "most people" would do what they find convenient. The way I get ready to print (and this is what I suggest to all my clients) is to set off a bed heat (takes, as you know, a fair while), and when it's up to temp then set off a hot end heat. When the hot end is up to temp change filament then and set the print job off.
If you are happy to go ahead and run your heater and have your hot-end at full temp while waiting for the bed to heat, go ahead. It's your printer.....

All I'm giving you is advice. If you don't like it feel free to ignore it.

Ghostkeeper, 2019-04-11:
> Can we add an option to change the default heating sequence?

There is. The printer's start g-code is exactly meant for this.

> Also I thought I was going crazy, but I did a google search of '3d printing heat stress filament' to see if this is really some phenomenon, and apparently it isn't. So maybe you can educate me on this.

We typically call this "filament degradation" in my office and it's the leading cause for clogged nozzles, at least in Ultimaker's printers. Prolonged heating makes the material brittle and oxidises it slowly, leaving a lump of coal in your nozzle.

closed 2019-04-11 by Ghostkeeper