Ultimaker/Cura #5586: Text rendering glitches on hi-dpi in settings dialogs

mqudsi, 2019-04-06:
With Cura 4.0 running on Windows 10 64-bit, there are odd issues rendering the text with letters randomly appearing much smaller than their neighbors. I'm presuming this is related to the hi dpi mode.

Here's an example:

[image]

Another possibility is that the selected fontface doesn't contain the letters and a different fontface is being used as a fallback which has a different x metric, but I don't think that's the case since in some cases (eg "zoom") the same letter is rendered once OK and once not.

xkero, 2019-04-07:
I also get some weird font rendering issues on my Linux hi-dpi system in just the settings dialog. Issue isn't present on my other non-hi-dpi Linux system.
[image]

mqudsi, 2019-04-07:
@xkero that looks like an issue with fixed size labels, their heights should rather be dynamically set/allowed to flow so that it may be dynamically computed at runtime, which means the container itself needs to be a flow container rather than a fixed-pixel widget.

rburema, 2019-04-10:
These dialogs are still in the 'old' qt controls 1 style. We'll be slowly replacing those kinds of dialogs anyway, if only to fit in with the rest of the interface. We'll take a closer look at if anything is rendered correctly when we (eventually) do that.

Ghostkeeper, 2019-04-11:
These issues are caused by the text positions not being aligned to whole pixels. That causes some weird anti-alias issues. In your case, it's probably because your screen has a pixel scale factor of 1.25 or something similarly uncommon.

I think that we'll postpone fixing this in the preferences dialogue until we've completed the redesign.