Ultimaker/Cura #5588: Upgrading CR-10 firmware

Ex-Pilot, 2019-04-07:
Hi.

I would like to update my CR-10 firmware, but as soon as I open Cura V4 my printer just starts resetting every 2 - 3 second and Cura writes there is a(n) communication problem.

Any ideas?

Liger0, 2019-04-07:
Why do you want to update a firmware using cura?

Ex-Pilot, 2019-04-07:
I can't find any other way to do this.

Liger0, 2019-04-07:
> 
> 
> I can't find any other way to do this.

?
Cura doesn't have anything to do with 3rd party firmware upgrades. 
Search a guide for flashing a bootloader (you'll need hardware), and then to flash marlin on it.

Ex-Pilot, 2019-04-07:
Cura has the possibility to "Update Firmware" so I thought it was able to do so?!

Liger0, 2019-04-07:
[](https://ultimaker.com/en/resources/52012-update-firmware)

Ex-Pilot, 2019-04-07:
That's what I tried, but my printer just starts resetting every 2 - 3 second and Cura writes "there is a(n) communication problem".

Liger0, 2019-04-07:
Because that's only for ultimaker machines..

Ex-Pilot, 2019-04-07:
Why is this option available for Creality if it's only for Ultimaker?!

fieldOfView, 2019-04-07:
> Because that's only for ultimaker machines..

That's not true. What is true is that it has only been _tested_ with Ultimaker machines (and the only other printer I have, which is a Blackbelt 3d printer). But it should work with any printer that is based on/behaves like an Arduino (Mega).

@Ex-Pilot, is your Creality otherwise working with Cura? Is it detected and can you print to it?

In the background, Cura uses a program called avrdude, which is also used by the Arduino IDE.

Had you filled in the issue template (... why didn't you?), it would have asked you for logs. Without logs there is not much I can do except speculate: It sounds like when the firmware upgrade process starts, the printer stops being a serial port, which somehow aborts the firmware upgrade process, which makes the printer reset and become a serial port again, at which point the firmware upgrade process is restarted, etc. This goes on until avrdude is done trying.

Liger0, 2019-04-07:
There's not way Cura may update a cr-10 without a bootloader.

fieldOfView, 2019-04-08:
No, if there is no bootloader, the upgrade will not work with Cura 2 and newer. At the risk of sounding unhelpfully pedantic, the function is “upgrade firmware”, not “install firmware”.

You could try an old version of Cura (15.x), which is more forgiving.

Ex-Pilot, 2019-04-10:
You all more or less mention Arduino or Arduino compatible. This tells me it's based on Atmel microcontrollers.

I have Atmel Studio 7.0 and Atmel STK500. Will this help? I also have Arduino 1.8.5 and some Arduino Nano boards (Atmel MEGA 328) based.

I wonder if it would possible to use my STK500 to flash the CR10 mainboard with the .HEX file from Creality?!

Ghostkeeper, 2019-04-11:
> Because that's only for ultimaker machines..

We can put any firmware in there that we could theoretically connect with via USB. We currently also have a Witbox firmware. It also supports uploading custom .hex files which is what Ex-Pilot is probably doing.

> You all more or less mention Arduino or Arduino compatible. This tells me it's based on Atmel microcontrollers.
> I have Atmel Studio 7.0 and Atmel STK500. Will this help? I also have Arduino 1.8.5 and some Arduino Nano boards (Atmel MEGA 328) based.

That Atmel thing is your chip, which can be something completely different and doesn't say much. We're using the Arduino drivers to send the firmware over from Cura, so it should function the same as if you're using the Arduio IDE. That's why we're talking about Arduino-compatible behaviour.

If Cura doesn't work for you, I'd suggest using Pronterface or the Legacy version of Cura (15.04 or earlier) as fieldOfView mentioned. For the rest, I don't think this is really a concern of Cura to make it work. Maybe the firmware itself is wrong and causes the printer to reset continuously?

closed 2020-06-20 by no-response[bot]