Ultimaker/Cura #5589: fail to do a pause after change a wire using usb

Josiko85, 2019-04-07:
**Application Version**
<4>

**Platform**

**Printer**
<ende3>

**Steps to Reproduce**
< pause at heigth>

**Actual Results**
<>

**Expected results**
<>

**Additional Information**
<only work belt if i use a microusb but when i use a usb wire never wait my to do a change and the work continue.>

rburema, 2019-04-10:
I understood this: If you save anything to a (solid-state) USB-disk, pause-at-height works, but if you print via USB-cable connected from your PC to your printer it won't pause. Is this correct?

Josiko85, 2019-04-10:
Yes. If i use usb-cable. The printer do all.( Move to the point 0;0 forma example. Retrac. And in this moment dosen't do the pause. Charge the wire and continue the process.

rburema, 2019-04-10:
Could you send the logs (see [here](https://github.com/Ultimaker/Cura/wiki/Cura-Preferences-and-Settings-Locations) for the location of the logs) and the .gcode file that is produced when saved to disk? (You may have to zip it to satisfy github.)

rburema, 2019-04-10:
Never mind, something about this jogged my memory:

This will be fixed by the fix for #5410 (since pause-at-height is a post-processing step, and the USB plugin previously, and in 4.0 as well, ignored post-processing scripts) _which means it'll be fixed in 4.1_, which makes this issue a duplicate.

Thanks for the report anyway!

closed 2019-04-10 by rburema