Ultimaker/Cura #5590: "One at a time" bounding box can't extend beyond build plate/bed

xkero, 2019-04-07:
**Application Version**
Cura 4.0.0

**Platform**
Linux

**Printer**
Creality Ender 3

**Steps to Reproduce**
Add an object, select "One at a time" for Print Sequence. Try to move object towards build plate/bed edge.

**Actual Results**
Cura tells me it can't slice the object.

**Expected results**
Can slice and print so long as object itself is within allowed build volume, like "All at once" allows.

**Additional Information**
[image]
**All at once**
Object close to edge and can slice, expected behaviour.

[image]
**One at a time**
Object close to edge and can **not** slice, unexpected behaviour.

[image]
**One at a time**
Object moved away from edge and can now slice, bad workaround as this means additional objects must be placed closer together which increases chance of collisions and other issues.

nallath, 2019-04-08:
Cura is completely correct in telling you this. It's likely that the ender 3 simply doesn't have it's machine properties set correctly in Cura (we completely depend on contributions for that as we can't test that without having a machine).

I do know they are set correctly for the UMO for instance, so if you look at both situations you can see the following happening;
[image]
[image]

In the case of all at once, we put the area of the brim at the edges, since multiple objects can be close together which merges their brims. This isn't the case for one at a time, as then we can't merge the brims (which is why the area around the model gets bigger).

xkero, 2019-04-08:
Yes I just added an Ultimaker Original and switched to that in Cura and now the correct behaviour is shown like in your pictures. How can I change the properties for the Creality Ender 3 to fix this? All I can find in Cura is the gantry size settings which seem to have no effect.

nallath, 2019-04-08:
You can set the bounding box of the head in the machine actions, but this only allows you to set a square head size. If you want to set a different shape, you need to modify the configuration definition file of the printer.

xkero, 2019-04-08:
Okay so changing the X/Y min/max values under "Printhead Settings" all to 1 shrinks the bounding box so I can have objects much closer to the edges, but I still don't agree that this is correct behaviour. Cura should only concern itself when the bounding box (which only appears when "One at a time" is selected) overlaps with another objects bounding box. This bounding box extending beyond the print plate/bed should not be an issue as demonstrated by Cura being happy to slice & print when selecting "All at once". Basically this feature in it's current state for the Ender 3 printer at least actually encourages print collision as you have to move objects towards the centre and closer together with the only workaround to effectively be to disable it.

TL;DR: Cura considers the void outside of the build plate/bed to be an obstacle, but only for the Ender 3 and only when "One at a time" is selected, all other tested configurations Cura acts rationally.

xkero, 2019-04-08:
Ahah! I found more evidence to show this is a bug. It's to do with the buildplate stl, I noticed that the Ultimaker one the build plate model extends beyond the actual build volume, but for the Ender 3 it doesn't as that printers build plate is just floating and has no enclosure.

I extended the "creality_ender3_platform.stl" build plate model and now Cura allows objects up to the edge of the build plate like it should do when "One at a time" is selected. So Cura is considering area outside of the build plate to be an obstacle which is not correct as the print head can not physically collide with areas it can't even reach. Again this is correctly handled when "All at once" is selected. It only seems to work correctly for the Ultimaker printer as the build plate model is larger than the build volume.

[image]

nallath, 2019-04-09:
You're misunderstanding what print one at a time does. Because you have platform adhesion, cura shouldn't let you print that close to the edge. Because it's not just the 3d model that is being printed, the brim / raft / whatever is also being printed. That's what is preventing your object from being placed right next to the edge.

xkero, 2019-04-09:
As I've already explained I have platform adhesion set to **none**, you can even see this in the screen shots I've submitted (it's the setting third up from Print Sequence). Also even if that was the case why wouldn't that also be a concern when set to All at once? The brim/whatever doesn't change when changing the Print Sequence setting.

Ghostkeeper, 2019-04-11:
Setting build plate adhesion to None, disabling Avoid Objects When Traveling, infill wipe distance set to 0, support horizontal offset set to 0, horizontal offset set to 0, travel avoid support disabled, no draft shield and no ooze shield, should result in allowing your object to be placed exactly on the edge.

Ghostkeeper, 2019-09-02:
Hmm, I think this is a matter of the printer configuration though. The Ender 3 specifies this as the shape of its print head:

        "machine_head_polygon": { "default_value": [
                [-26, 34],
                [-26, -32],
                [22, -32],
                [22, 34]
            ]
        },
        "machine_head_with_fans_polygon": { "default_value": [
                [-26, 34],
                [-26, -32],
                [32, -32],
                [32, 34]
            ]
        },

I'm not entirely sure why we're still using the print head without fans for the model's collision area itself, but it should be WAY smaller than the print head with fans. Basically just the nozzle's size of where it could hit other models while printing, like 2mm diameter. There's a lot of confusion about these two settings and how they relate to each other though so I hope we can find a way to clear that up.

Can you try setting the Machine Head Polygon to `[[-1,1], [-1,-1], [1,-1], [1,1]]`? You can change it using the Machine Settings plug-in in the marketplace (or by editing your printer's profile in Program Files).

dschraitle, 2019-09-02:
Ah, yes, thank you! Changing the Machine Head Polygon shrinks the previously off-center box to just a small perimeter and now I'm able to place models correctly.

Ghostkeeper, 2019-09-05:
If you changed that, is it possible to place objects so close together that the print head collides with previously printed models? If collisions are still impossible, we can change that in the source files for a next release.

dschraitle, 2019-09-05:
Yes, placing objects close enough such that when the larger bounding box (I'm guessing it's the one based on `machine_head_with_fans_polygon`) intersects with another model, Cura is unable to slice.

Ghostkeeper, 2019-09-06:
What I mean is to place the objects far enough apart that Cura is JUST allowing you to slice (so much closer than previously) and then printing that. See if the print head collides with the previously printed parts (so be ready at the power switch). The aim is to prevent damage to the printer.

dschraitle, 2019-09-09:
Just tried it, and there was no print head collision.

Ghostkeeper, 2019-09-09:
I've applied that change then.

closed 2019-09-09 by Ghostkeeper

nipunawg, 2019-12-05:
Tevo Tarantula owner here. I have ridiculous bounding box in the default settings too. It still hasn't been changed. maybe its the same for other printers as well?

dschraitle, 2019-12-05:
The change was made only for a certain type of printer, other printer types
would require the same change to be made to their base definition files. If
you look at the linked commit you can see the location of the file for that
printer that was changed. Modifying the file locally and restarting Cura
works for me.

nipunawg, 2019-12-05:
Yeah I fixed it manually for myself. Just letting others know.

Ghostkeeper, 2019-12-06:
If there's an improvement to the printer definitions, that you've applied and tested, we can publish it with the next Cura release if you want.

commit a43de191 (2019-09-09): Reduce size of print head without fans for one-at-a-time mode

This allows you to place models more compactly.

According to the discussion at #5590, this was tested to work well.

Fixes #5590.
commit 8c817e04 (2019-09-09): Reduce size of print head without fans for one-at-a-time mode

This allows you to place models more compactly.

According to the discussion at #5590, this was tested to work well.

Fixes #5590.