Ultimaker/Cura #5591: GCODE viewer cannot load the gcode just sliced

chputney, 2019-04-07:
Cura 4.0.0
Using  Windows 7

For some reason the gcode viewer is not working.  After a model is sliced, save the sliced model and delete it from the platform.  Now load the saved gcode file.

After parsing the file to 99%, then an error is displayed "Invalid File Failed to load C:\xxxxx"

My 3mf project file is attached with a simple cube that shows the error.

[Cube.zip](https://github.com/Ultimaker/Cura/files/3051830/Cube.zip)

nallath, 2019-04-08:
Could you share the logs? They contain some extra information as to what went wrong (I can load g-code just fine)

chputney, 2019-04-08:
Here is the log file.  The last thing I did was to try to load a gcode file.
[cura.zip](https://github.com/Ultimaker/Cura/files/3054041/cura.zip)

nallath, 2019-04-08:
You seem to have incorrectly changed something. I guess somewhere in your start / stop / end g-code, there is a tag with {TRAVEL_SPEED} (which should be {travel_speed}). When writing, this tag is not replaced, so it remains in the g-code (because it's invalid).
Cura subsequently chokes on trying to load it because it's invalid g-code.

chputney, 2019-04-08:
I have not changed any start end gcode from the printer definition.  

I did some more checking and it happens with the printer "Wanhao Duplicator i3 Plus".  I used to use a custom printer, but this is now a printer in the built in list.  When I use my old custom printer, there is no problem with loading gcode just sliced from the stl model.  I used a printer called "Ultimaker Original" and again there was no problem.  When I select the printer "Wanhao Duplicator i3 Plus" I have the issue.  

I see "{travel_speed}" in the end gcode and now I believe that this is because the printer definition has this typo.  I replaced "{travel_speed}" with "50" and I can now view the gcode files.

I then deleted the printer and re-installed it.  Now the ending gcode is "{speed_travel}" and the gcode produced from an stl model and saved will correctly load.

I wonder if this was an error in the original definitions in 3.6 and when I upgraded to 4.0 was carried over.  Anyway, it looks like the problem is resolved.  This issue can be closed.

fieldOfView, 2019-04-08:
> there is a tag with {TRAVEL_SPEED} (which should be {travel_speed}

It should be `{speed_travel}`

nallath, 2019-04-09:
AH right. These were generated from the old Cura, which had the tags as {travel_speed}. I've fixed them in master.

commit 87c4c032 (2019-04-09): Fix incorrect tag in start g-code

Fixes #5591
closed 2019-04-09 by nallath