Ultimaker/Cura #5593: [4.0] GUI error - OpenGL context broken

tingspain, 2019-04-07:
**Application Version**
Cura 4.0 Release

**Platform**
Windows 7 64bits

**Printer**
no aplicable 

**Steps to Reproduce**
Basically, I have the problem as soon as I loaded an object to be printed. However, sometimes this issue doesn't show immediately. In order to force to reproduce the issue, I follow the following steps: 
1.- Open any STL
2.- click anywhere on the top menu, to open a dropdown menu
3.- and the OpenGL context of the Scene View should disappear

**Check the Youtube video attached below for a better understanding of the issue ! ! !**

**Actual Results**
All the GUI disappears.

**Expected results**
The GUI should be visible during all the time

**Additional Information**
<iframe width="560" height="315" src="https://www.youtube.com/embed/Fl74eiVt9rk" frameborder="0" allow="accelerometer; autoplay; encrypted-media; gyroscope; picture-in-picture" allowfullscreen></iframe>

nallath, 2019-04-08:
Could you share your logs and double check that your drivers are up to date?

rburema, 2019-04-10:
If it still occurs with updated drivers: What graphics card are you using? Which version of OpenGL is used? (This can be found in the logs.)

Ghostkeeper, 2019-04-12:
Weird how it works fine at first until you open an additional menu. Maybe it's running out of GPU memory? Can you try going into the preferences screen and selecting compatibility mode for layer view?

tingspain, 2019-04-12:
Hello, 

First of all, I am sorry for my late reply but I have been busy. 

**UPDATE: ISSUE HAS BEEN FIXED:**

I am running Cura on a laptop with an integrated Intel video card and a dedicated Nvidia card. I thought that I added Cura in the high graphics card programs list to automatically run Cura with the nVidia card, however, it seems I didn't save it properly and I was running Cura with the intel card.  
But I added Cura properly to the high graphics programs list and the issue disappeared. 

PD: I have the latest drivers for the intel and the Nvidia cards

closed 2019-04-12 by Ghostkeeper