Ultimaker/Cura #5598: Tree support generates dots in 3.6 & 4.0

batesman1, 2019-04-08:
[Model_and_project_file.zip](https://github.com/Ultimaker/Cura/files/3055621/Model_and_project_file.zip)
[Model_and_project_file.zip](https://github.com/Ultimaker/Cura/files/3055624/Model_and_project_file.zip)

**Application Version**
3.6 & 4.0

**Platform**
Windows 7

**Printer**
Creality Ender2 and 3

**Steps to Reproduce**
As the issue appears to lie with this specific model, just slice this model with tree support enabled. 

**Actual Results**
Small dots appear and are printed. The number of dots goes up as the tree support branch distance is increased or decreased.

**Expected results**
I expect proper tree supports to be generated without all the extra dots

**Additional Information**
The profile used doesn't seem to make any difference as I have tried with custom profiles as well as .2 and .15 standard profiles as well. In the attached .3mf file, the tree support branch distance is set to 3 to shorten slicing time and support is being generated only off the buildplate.

Liger0, 2019-04-08:
I noticed a very high amount of random dots using 4.0 too. I have also a doubel extrusion machine and the top and bottom layers use the second extruder (PVA), and the dots are traited like interface layers, so they use the second extruder.

batesman1, 2019-04-08:
Another user had this issue using 3.2.1 on a separate model.

https://community.ultimaker.com/topic/27473-tree-support-dots/

batesman1, 2019-04-08:
[image]

rburema, 2019-04-10:
Could _maybe_ be related to an internal ticket we have, CURA-6424, _however_ the behaviour here is different. _I also notice that on layer 6, the support seems to skip a layer._ I think we'll need to investigate.

batesman1, 2019-04-11:
So, I went back into tinkercad and adjusted the model. Specifically, I changed the radius which affects how round the corners (internal and external) are. By reducing the radius and making the object more square, it now slices correctly.
[baffle redux.zip](https://github.com/Ultimaker/Cura/files/3066762/baffle.redux.zip)
[image]

Project file, revised stl

batesman1, 2019-04-24:
Different model. Same problem. The issue appears to be tied to the tree support branch distance.

[image]

Now observe the same model with only the tree support branch distance changed from 1 to 2.

[image]

and from 2 to 3

[image]

batesman1, 2019-04-24:
By changing the tree support branch diameter, I can make the tree supports do this, even when there is no roof to support.

[image]

batesman1, 2019-04-24:
Here is the project file and stl
[model_ project.zip](https://github.com/Ultimaker/Cura/files/3110678/model_.project.zip)

batesman1, 2019-04-24:
Or perhaps the issue stems from models that have rounded internal structure? This is the same model, flipped upside down.

[image]

Ghostkeeper, 2019-04-24:
I've seen it happen quite often with round holes. I think that for some reason it's ignoring the holes in an object when determining the area that should be supported with tree branches.

Ghostkeeper, 2019-06-12:
This got fixed by @schneider007 through this pull request: https://github.com/Ultimaker/CuraEngine/pull/937

closed 2019-06-12 by Ghostkeeper