Ultimaker/Cura #5599: [4.0] Scaling UI issues after rotation

cp2004, 2019-04-08:
**Application Version**
4.0
**Platform**
Windows 10, 64bit
**Printer**
Custom FDM Printer (My Tronxy)
**Steps to Reproduce**

Import an object into Cura, rotate it in any axis, and then scale with the X/Y/Z arrows on the model
**Actual Results**

(For model meshes) The scale measurements are right, however the scale factor percentage is in the wrong box. 
(For support blockers) The scale arrows scale in the wrong direction. For X it scales Y, for  Y it scales Z. I think it changes depending on the rotation though.
**Expected results**

All axis scale in the right direction and the UI updates correctly
**Additional Information**

[scalebugreport.zip](https://github.com/Ultimaker/Cura/files/3056251/scalebugreport.zip)

rburema, 2019-04-10:
I can confirm that this happens in current master as well. I didn't need to rotate the support (on an already rotated model) to get the wrong behaviour though. We'll very probably make a ticket for it soon.

Ghostkeeper, 2019-04-12:
Duplicate of https://github.com/Ultimaker/Cura/issues/4136

closed 2019-04-12 by Ghostkeeper