Ultimaker/Cura #5600: [4.0.0] Cura takes a long time to load models when USB-to-RS232 devices are plugged in

petemate, 2019-04-08:
Hi, I have several RS232 devices connected to my computer using HL-340 USB-to-RS232 converters. However, Cura is extremely slow when loading models(STL files) if these converters are plugged in when Cura starts. If I unplug the converters before starting Cura, I can re-plug the converters and Cura will continue to load models just fine. From the stderr.log, it appears that Cura is just continuously running AutoDetectBaudJob on all ports. After some time, the model loads.

In the attached log file, I open cura at 00:32:00 and open a model at 00:32:20(thats when I click the "Open" button in the dialog box). But the model doesn't appear until approx 00:35:35, which matches with the timestamp 00:35:34,162 from the log file.

**Application Version**
Cura 4.0.0 App image

**Platform**
Linux <computer name> 3.16.0-8-amd64 #1 SMP Debian 3.16.64-2 (2019-04-01) x86_64 GNU/Linux

**Printer**
The printer selected is an Ender 3, but I doubt it has anything to do with this issue, so I haven't attached the project file.  

**Steps to Reproduce**
To reproduce, plug in a HL-230 USB-to-RS232 converter, start Cura and load a model. 

**Actual Results**
After approx. 3 minutes, the model loads. This coincides with the end of AutoDetectBaudJob,

**Expected results**
The model should load immediately, as if the USB-to-RS232 converters were not plugged in.

**Additional Information**
None. 

[stderr.log](https://github.com/Ultimaker/Cura/files/3056492/stderr.log)

rburema, 2019-04-10:
Cura is trying to see if there are printers/devices connected to those ports, so in a way this is expected behaviour. I remember that we had a discussion about something similar with the team before, so just to get it clear I'll ask again.

petemate, 2019-04-10:
Yeah, its just "too persistent" when searching for other devices. Thats the problem :)

rburema, 2019-04-10:
Yeah I got that, just trying to wreck my brain on what was the problem last time this (or something suspiciously similar) came up :-)
(Hence the ask around statement.)

Ghostkeeper, 2019-04-12:
I see what's happening here. When rburema described it, I thought this was about the boot time but it's actually about loading a file.

The problem here is that the USB printer detection is a separate asynchronous "job" which gets executed on a separate thread in Uranium. However Uranium has a queue of jobs to execute and won't start the next job until the previous one is completed. While it's properly multithreading alongside the GUI and the main thread, it's not multithreading alongside other jobs.

rednaz1337, 2021-01-17:
Same problem here on a ThinkPad W500 but without any USB to Serial adapters plugged in. Some internal devices on this Laptop have a `/dev/ttyACMx` device file. Cura probably thinks these are printers. In reality, they are modems.

ant-7802, 2024-06-09:
please fix this :pray: