Ultimaker/Cura #5601: [4.0.0] macOs crash on Layer view

ludiazv, 2019-04-09:
**Application Version**
4.0.0

**Platform**
macOs (El capitan)

**Printer**
N/A

**Steps to Reproduce**
Slice model and select 

**Actual Results**
crash

**Expected results**
layer interface

**Additional Information**
Crash log:
````
Process:               cura [39095]
Path:                  /Applications/Ultimaker Cura.app/Contents/MacOS/cura
Identifier:            nl.ultimaker.cura
Version:               4.0 (4.0.0)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           cura [39095]
User ID:               501

Date/Time:             2019-04-09 09:40:43.198 +0200
OS Version:            Mac OS X 10.11.6 (15G22010)
Report Version:        11
Anonymous UUID:        F35E4116-36B6-3609-05E4-B27FEC496EE1

Time Awake Since Boot: 3200000 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000020
Exception Note:        EXC_CORPSE_NOTIFY

VM Regions Near 0x20:
--> 
    __TEXT                 000000010c1d4000-000000010c37c000 [ 1696K] r-x/rwx SM=COW  /Applications/Ultimaker Cura.app/Contents/MacOS/cura

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff896708ea __kill + 10
1   libsystem_platform.dylib      	0x00007fff9632052a _sigtramp + 26
2   ???                           	000000000000000000 0 + 0
3   com.apple.driver.AppleIntelSKLGraphicsGLDriver	0x000000012030a12f (anonymous namespace)::AlwaysInliner::getInlineCost(llvm::CallSite) + 111
4   com.apple.driver.AppleIntelSKLGraphicsGLDriver	0x000000012030a713 llvm::Inliner::shouldInline(llvm::CallSite) + 35
5   com.apple.driver.AppleIntelSKLGraphicsGLDriver	0x000000012030bb60 llvm::Inliner::runOnSCC(llvm::CallGraphSCC&) + 2160
6   com.apple.driver.AppleIntelSKLGraphicsGLDriver	0x00000001202ffcbd (anonymous namespace)::CGPassManager::runOnModule(llvm::Module&) + 1165
7   com.apple.driver.AppleIntelSKLGraphicsGLDriver	0x00000001201530ae llvm::legacy::PassManagerImpl::run(llvm::Module&) + 926
8   com.apple.driver.AppleIntelSKLGraphicsGLDriver	0x000000011fd1e435 IGC::UnifyIROGL(IGC::CodeGenContext*) + 885
9   com.apple.driver.AppleIntelSKLGraphicsGLDriver	0x000000011fc029fa void ILAdapter::CreateShader<IGC::VertexShaderContext, USC::SCompilerOutputVertexShaderOpenGL_Gen9>(USC::s_shader*, void*, USC::SCompilerInputStateCommonOpenGL*, USC::SCompilerOutputVertexShaderOpenGL_Gen9*&) + 666
10  com.apple.driver.AppleIntelSKLGraphicsGLDriver	0x000000011fbda578 USC::CompileVertexShaderOpenGL(void*, USC::OPTIMIZER_LEVEL, USC::SCompilerInputStateVertexShaderOpenGL const*, USC::SCompilerOutputVertexShaderOpenGL_Gen9*&) + 40
11  com.apple.driver.AppleIntelSKLGraphicsGLDriver	0x000000011fb7ae82 USC::IM_Shader::Compile(void*, GFXCORE_FAMILY, void*) + 816
12  com.apple.driver.AppleIntelSKLGraphicsGLDriver	0x000000011fb95c04 intelUpdateCtxSysProgramCommon(GLDContextRec*, GLDProgramRec*, GLDPipelineProgramRec*, void*, ShaderInfoCacheRec*, unsigned int) + 3683
13  com.apple.driver.AppleIntelSKLGraphicsGLDriver	0x000000011fb93314 gpusLoadCurrentPipelinePrograms + 548
14  com.apple.driver.AppleIntelSKLGraphicsGLDriver	0x000000011fb90e3e updateDispatch + 1710
15  GLEngine                      	0x00007fff88858037 gleDoDrawDispatchCoreGL3 + 533
16  GLEngine                      	0x00007fff888087a8 gleDrawArraysOrElements_Entries_Body + 128
17  GLEngine                      	0x00007fff88801ae8 glDrawElements_GL3Exec + 214
18  PyQt5._QOpenGLFunctions_4_1_Core.so	0x00000001167502f2 meth_QOpenGLFunctions_4_1_Core_glDrawElements(_object*, _object*) + 178
19  nl.ultimaker.cura             	0x000000010c228f95 PyCFunction_Call + 165
20  nl.ultimaker.cura             	0x000000010c2a63c0 PyEval_EvalFrameEx + 25584
21  nl.ultimaker.cura             	0x000000010c2ab3a2 0x10c1d4000 + 881570
22  nl.ultimaker.cura             	0x000000010c2a5d0f PyEval_EvalFrameEx + 23871
23  nl.ultimaker.cura             	0x000000010c2ab3a2 0x10c1d4000 + 881570
24  nl.ultimaker.cura             	0x000000010c2a5d0f PyEval_EvalFrameEx + 23871
25  nl.ultimaker.cura             	0x000000010c2ab3a2 0x10c1d4000 + 881570
26  nl.ultimaker.cura             	0x000000010c2a5d0f PyEval_EvalFrameEx + 23871
27  nl.ultimaker.cura             	0x000000010c2ab3a2 0x10c1d4000 + 881570
28  nl.ultimaker.cura             	0x000000010c2a5d0f PyEval_EvalFrameEx + 23871
29  nl.ultimaker.cura             	0x000000010c2aabb2 0x10c1d4000 + 879538
30  nl.ultimaker.cura             	0x000000010c29ffae PyEval_EvalCodeEx + 78
31  nl.ultimaker.cura             	0x000000010c20952d 0x10c1d4000 + 218413
32  nl.ultimaker.cura             	0x000000010c1dfec5 PyObject_Call + 101
33  nl.ultimaker.cura             	0x000000010c1f5c0d 0x10c1d4000 + 138253
34  nl.ultimaker.cura             	0x000000010c1dfec5 PyObject_Call + 101
35  nl.ultimaker.cura             	0x000000010c2a9b25 PyEval_CallObjectWithKeywords + 165
36  PyQt5.QtCore.so               	0x0000000117274238 PyQtSlot::call(_object*, _object*) const + 40
37  PyQt5.QtCore.so               	0x0000000117274136 PyQtSlot::invoke(void**, _object*, void*, bool) const + 374
38  PyQt5.QtCore.so               	0x0000000117274b28 PyQtSlotProxy::unislot(void**) + 88
39  PyQt5.QtCore.so               	0x0000000117274a9a PyQtSlotProxy::qt_metacall(QMetaObject::Call, int, void**) + 58
40  libQt5Core.5.dylib            	0x0000000117637437 QMetaObject::activate(QObject*, int, int, void**) + 1959
41  libQt5Quick.5.dylib           	0x000000011a6cfdb4 QQuickWindowPrivate::renderSceneGraph(QSize const&) + 116
42  libQt5Quick.5.dylib           	0x000000011a661abd 0x11a5b7000 +

```

rburema, 2019-04-10:
I assume your graphic drivers are up to date? What're the graphics 

Some speculation:
Since I see `llvm` in the log up there, which is used in the software/mesa rendering pipeline, I suspect:
This _may_ be related to the issue where if we ask for a OpenGL 4.x context on Macs without a compatible graphics card/driver, it'll just give us the software-rendering one instead declining so we can use the (probably supported) 2.0 fallback context. Hypothesis: Maybe it needs a lot of memory for those operations which it couldn't allocate?

In any case, could you also send us the Cura logs? They're in `$User/Library/Application\ Support/Cura/$CURA_VERSION/` for Mac.

fieldOfView, 2019-04-11:
```llvm``` is a compiler. Other drivers could have been compiled with the same compiler.

@ludiazv, what GPU do you use?

rburema, 2019-04-11:
I know, I just also know that mesa makes extensive use of the _libraries_ supplied by llvm as well. But you're right, nVidea or the others could do the same.

ludiazv, 2019-04-11:
Thanks I will try post the log later today.
The GPU integrated intel HD series. I had in the past no problems whatshoever with this gpu. I use other programs such Kicad or Fussion360 that use OpenGL with acceleration working great. I don't think is a hw problem.

Ghostkeeper, 2019-04-12:
Just to be sure and provide a possible workaround, could you go into the preferences and select compatibility mode for layer view?

fieldOfView, 2019-04-12:
The "integrated intel HD series" is a rather big family of gpus, with wildly varying capabilities. What CPU do you have? Note that "Intel Core i5" does not say much, because again that's a whole series of product with many variations. What processor speed, and what year is your model?

ludiazv, 2019-04-13:
with compatibility layer enabled works! 👍

rburema, 2020-02-13:
This came up again in a slightly different context and is now also (probably) linked to internal ticket CURA-7202