Ultimaker/Cura #5602: Support Blocker: Scaling Axis rotated

krz4tw, 2019-04-09:
**Application Version**
4.0.0

**Platform**
Win 10 x64

**Printer**
i3 Mega

**Steps to Reproduce**
Import an STL
Rotate the the STL 90 degrees
Add Support Blocker
Uncheck Uniform Scaling for Support Blocker
Scale Support blocker

**Actual Results**
If you want to scale your support blocker after you´ve rotated the STL file (only the STL, not the Blocker itself!) the axis for scaling the blocker get messed up. If I grab the blue Z handle and drag, the blocker scales in the Y direction. If I grab the green Y handle the Z direction scales.
I havent tested if it depends around which axis and how much you´ve rotated your STL but I suspect the X-axis (that works as expected) was the one I rotated the SL around in the beginning.

**Expected results**
Scale in the right direction

Thanks

rburema, 2019-04-10:
Seems to be related to/duplicate of #5599, can you confirm?

Ghostkeeper, 2019-04-12:
Yeah, with uniform scaling disabled and such the percentages are wrong, yada yada. Duplicate of https://github.com/Ultimaker/Cura/issues/4136.

closed 2019-04-12 by Ghostkeeper