Ultimaker/Cura #5603: Saving Project Files: Support Blocker Lifts Part

krz4tw, 2019-04-09:
**Application Version**
4.0.0

**Platform**
Win7 x64

**Printer**
i3Mega

**Steps to Reproduce**
Import STL Model
Add Support Blocker
Move and and Scale it
Save as mf project file
Close and Reopen Cura
Open project file

**Actual Results**
After opening the saved project file, the model and the supportblocker get lifted. The part and the blockers relative positions to one another stay the same but where before saving, the STL was laying flat on the buildplate, Cura has now lifted model and blocker so that the bottom of the blocker is laying flat on the build plate.

**Expected results**
Don´t move parts

Here a picture after saving a dummy cube project and reoping it:
[image]

***
It just dawned on my when I looked on the picture just now that it also rotates the support blocker.
I had rotated the STL before adding the blocker. I strongly suspect this is connected to #5602

Thanks

rburema, 2019-04-10:
... and #5599 as well, then.

krz4tw, 2019-04-10:
> ... and #5599 as well, then.

In reagards to the rotating support block certainly, but that issue was just something that came out of nowhere and I just realized when I looked at the screens I made to showcase the real issue.

This one is about the lifting of model and block primarily.
I did a few test and it seems that it only occurs if you´ve roted the STL beforehand though.

Thanks

Ghostkeeper, 2019-04-12:
Yeah this is a different issue from #4136 and its duplicates.

The problem is that we align the object on the build plate, regardless of whether it is a support blocker or not.

commit e8d2e070 (2019-04-15): Dont calculate bounding box for non printing meshes

Fixes #5603
closed 2019-04-15 by nallath