Ultimaker/Cura #5605: Disabling extruder changes support material

BagelOrb, 2019-04-10:
**Application Version**
4.0.0

**Printer**
UM3, S5

**Steps to Reproduce**
Load PVA in extruder 2
Enable support, with PVA
...
Disable the second extruder

**Actual Results**
Support is being printed with PLA

**Expected results**
No support is being printed

**Additional Information**
This is a UX issue. The question is what a normal user would expect. As a noob user I would think that PVA is the support, so disabling PVA would disable support.

rburema, 2019-04-10:
Hi @BagelOrb , I've put a sticky on the CCB so it'll be discussed tomorrow-morning :-)

BagelOrb, 2019-04-10:
I bet it will be a 'Won't fix'.

BagelOrb, 2019-04-10:
Oops

rburema, 2019-04-11:
So, the consensus was that, if you have a viable suggestion for what _should_ happen in this case instead, we'll consider implementing it.

BagelOrb, 2019-04-11:
I don't understand how that was not clear. When you disable the support extruder then the setting Enable Support should automatically become false.

fieldOfView, 2019-04-11:
Personally I would say that the current behavior is expected behavior.

rburema, 2019-04-11:
So then disabling and then renabling the extruder would have a side-effect?

fieldOfView, 2019-04-11:
Yes. But that does not change if disabling the support extruder disables support (Tim's expected result).

If you set the infill to be printed with the secondary extruder and disable the secondary extruder, should there be no infill? If you load a model, make it print with the secondary extruder, then disable the secondary extruder. Should that stop the model from being printed?

rburema, 2019-04-11:
Sorry, I should have tagged Tim, I was responding to his latest comment.

BagelOrb, 2019-04-11:
Ok never mind.

closed 2019-04-11 by BagelOrb