Ultimaker/Cura #5607: [4.0] Excluding materials

filipgoc, 2019-04-10:
**Application Version**

4.0
**Platform**

mac/windows...
**Printer**

Ultimaker Original
**Steps to Reproduce**

Try to select a material in Cura
**Actual Results**

A list of materials appears and includes Ultimaker and some Generic materials
[image]
**Expected results**

Here's my puzzle:
In ultimaker.def.json and in ultimaker_original.def.json, the only excluded materials are some generics and some ultimaker materials. 
[image]
[image]

Based on this, I would expect materials from other manufacturers to also appear unless explicitly excluded. Where are all the other materials being excluded? 

When I try the same with my printer (JellyBOX), all materials appear unless I explicitly exclude them. 

(This has been the case for a long time, even though even a longer time ago exclusion was the default and we used to list compatible materials...)

**Additional Information**

I'm quite curious. Thanks.

rburema, 2019-04-10:
Different causes are/could be to 'blame':
 - Other materials inherit from the 'generic' ones, otherwise the list of exclude materials would be almost as unmanagable as the include materials. If someone makes a new material that inherits from, say, 'generic_pva' we don't want to have to exclude it manually.
 - Supported nozzle sizes/other specs found in the material definition could be different (though I don't know/remember if we exclude based on that).
 - Unlikely, since they show up on your JellyBOX, but if anyone else reads this: materials are available in the Marketplace, Cura will need to restart before it has effect after installing.

filipgoc, 2019-04-11:
> Other materials inherit from the 'generic' ones, otherwise the list of exclude materials would be almost as unmanagable as the include materials. If someone makes a new material that inherits from, say, 'generic_pva' we don't want to have to exclude it manually.

True story, but it does not explain what I'm seeing. In fact, that would explain the opposite :-) 

> Supported nozzle sizes/other specs found in the material definition could be different (though I don't know/remember if we exclude based on that).

Of course! I did not realize that Ultimaker is quite alone in using 2.85mm diameter. That's the only reason why all those materials that show up for my JellyBOX do not show up for the Ultimakers. I re-tested it with some test materials and that's it.

Mystery solved. Thanks.

closed 2019-04-11 by filipgoc