Ultimaker/Cura #5608: fix temperature - Dual Printer

raphiiiii001, 2019-04-10:
Hey Guys,

I'm working on my own printer, so I have my own definitions and that's all. The only problem I still have is that when I create a GCODE there is always a problem with the temperature of the extruder. For example, whatever happens is that the standby temperature is always applied to the wrong extruder, so if extruder 1 is printing and extruder 2 is not, E1 is at standby temperature and E2 is at print temperature. And I have no idea how I can change that

So my question is, can I just fix temperatures, so that the extruder have constant, for example, 200 degrees and it does not change throughout the course of printing?

thank you for your help
kind regards 
R
:D

rburema, 2019-04-10:
Could you show us the definitions and/or what happens to the gcode? I think your proposed fix is just masking the problem. (And if you're not carefull to unload the other printcore yourself, that extruder will ooze all over the buildplate.)

P.S. We'd prefer it if you used the bug-report template that shows up when you make a new issue. It's there for a reason :-)

Ghostkeeper, 2019-04-12:
If you set the printing temperature to 200 in the start g-code, set the printing temperature to 200, the initial layer printing temperature to 200, the initial temperature to 200, the final temperature to 200 and the stand-by temperature to 200, for all extruders, it should always set your temperatures to 200.

raphiiiii001, 2019-04-16:
> If you set the printing temperature to 200 in the start g-code, set the printing temperature to 200, the initial layer printing temperature to 200, the initial temperature to 200, the final [...]

How exactly can I do that, I do not know exactly what you mean or how you think I should do that

Ghostkeeper, 2019-04-17:
1. Open the custom print settings screen (where all of the settings are displayed).
2. In the search field type the word "temperature".
3. Change all settings that you see to 200, except the ones where the build plate is mentioned.
4. Go to the second extruder tab.
5. Change all of those settings again to 200, except the build plate temperatures.