Ultimaker/Cura #5609: [4.0.0] Raft+Support causes model to float

TheGreatBert, 2019-04-11:
**Application Version**
4.0.0

**Platform**
Windows 10

**Printer**
Creality Ender 3

**Steps to Reproduce**
With a model that requires support that reached the build plate set build plate adhesion to raft.  the raft will be printed and then some layers of support before any part of the model that should touch the raft.  there is an air gap of 4 layers between the model and the raft.  If support is turned off the model touches the raft.  if raft is off an support is on the model touches the build plate.

**Actual Results**
Model does not sit on the raft

**Expected results**
Model should touch the raft in areas where it touches the raft when support is not included.

**Additional Information**
Here is the gap shown with raft and support on.  layer 4 is the top of the raft but the model does not start until layer 9.
[image]

here is the same model with support off.  the raft is still 4 thick but the model starts at layer 5.
[image]

support on and raft off.  model begins at layer 1.
[image]

support on and brim on model still begins at layer 1 and support is on top of the brim.
[image]
[image]

rburema, 2019-04-11:
I've tried to reproduce this, and it also happens when 'raft air gap' is set to 0 mm, and only happens if the model would otherwise also be printed with support (as opposed to just having support enabled). Probably a bug, but will discuss (hopefully) tomorrow.

Ghostkeeper, 2019-04-12:
I'm not able to reproduce this though. This is what I get when I set the air gap to 0:

[image]

The initial layer + air gap gets filled with support layers. This may result in a small gap at the bottom of the support if it's not an exact multiple. But for the rest there should be no more gaps if the air gap is 0 (for the correct extruder).

rburema, 2019-04-12:
@TheGreatBert  I think we figured it out: Could you look at what happens from the side on to that model? (The more exact the side view, the better.) What we could see is that the initial layer of the model is thicker than the support layers (as expected in this case), causing the suppor to show up in lower layers in the preview, even though the ticker 'initial' layer is deposited to the same physical z-level.

TheGreatBert, 2019-04-13:
@rburema I'm not sure I understand what you are asking.  My first pic id from the side.  What would you like different about it?

rburema, 2019-04-13:
@TheGreatBert Ah, forgot to mention: Please also set the 'Raft Air Gap' to 0mm?

TheGreatBert, 2019-04-16:
When I set air gap to 0mm it prints layers 5 and 6 as support and layer 7 as model, but it appears to remain at the same Z between layers 6 and 7 and just prints layer 7 thicker.  That should work ok.

[image]

[image]

Ghostkeeper, 2019-04-17:
So this seems to be working correctly then. The air gap is only applied to the model and not the support. And because the order in which layers appear is sorted by the Z coordinate of the top of the layer rather than the bottom, it appeared as if it was printed out of order while it really wasn't if you look closely enough.

closed 2019-04-17 by Ghostkeeper

JimMcCrank, 2021-10-19:
I am currently having this problem happen and it appears the more are suddenly having it happen to them. I will try setting my airgap but I did not touch this setting before and suddenly an extra gap was included in my sliced prints. Hopefully a reason for this is found but for now I will take your advice and set my airgap to 0. Thanks

arutz12, 2021-12-05:
OK, so when there is raft addhesion type and the model needs support touching the ground (the raft actually) the support layer comes always before the initial layer. Not both of them together. Therefore the first layer floats. Depending of the layer height even more layers of support are printed before the initial layer of the model. I tried the latest version of Cura as well, nothing has changed. Really annoying and I cannot remember when this problem appeared.
I have to print with brim or skirt if the first layer contains support. I tried several settings, printers, re-installed, tried new installation, nothing worked.
It seems that the initial layer of the support is handled differently when printed on a raft.