Ultimaker/Cura #5610: Cura 4.0 crashes when Nvidia GeForce video drivers are updated

tacomarocker, 2019-04-11:
**Application Version**
4.0.0

**Platform**
Windows 10

**Printer**
Ender 3 printer [Thingiverse file](https://www.thingiverse.com/thing:2531450)

**Steps to Reproduce**
1. Launch Cura 4.0.0
2. Add Files "Shelf1and3-Left-Vallejo" and "Shelf1and3-Right-Vallejo" (orientation rotated to 90)
3. Start the print (connected via USB)
4. Receive update for Nvidia video drivers
5. Run the update during a print

**Actual Results**
During the update of video drivers the screen flashes a few times while updating, during the update Cura crashed. Steam, a game, Chrome, and File Explorer were all open at the same time non crashed.

**Expected results**
Cura should remain open during a print and if interrupted by crash or power failure have a resume print by viewing a log of the current print status up until failure.

**Additional Information**
N/A

fieldOfView, 2019-04-11:
There is no way Cura can be expected to keep running if the rug is pulled from under it.

Don't do any driver updates while printing over USB. Better yet, don't use Windows for printing over USB, because it is prone to "receive updates" at unexpected times, and this will ruin your prints. 

Best advice I have is to install OctoPrint on a separate computer (can be as simple as a Raspberry Pi), and you will have a much more stable platform.

Ghostkeeper, 2019-04-12:
Best we could do is store the position where we were at regularly on the hard drive so that a print could resume after a power failure on the computer's side. I would not consider that to be entirely out of scope for Cura, but it's not something that we will be able to spend time on any time soon because Ultimaker's newest printers all work via SD cards or wifi.

Let's just say that PRs are welcome. However implementing a workflow for detecting that a previous print was in progress and never finished and then introducing a dialogue to continue and such could be a bit involved for a PR.

digitalfrost, 2022-07-27:
**Stale ticket.**
Can this be closed, since this refers to Cura 4.0 and we are now on 5.0  and the label is **Won't Fix/Do** ?

tacomarocker, 2022-07-27:
As fieldOfView pointed out it's best not to use USB while printing, I had just started printing at the time I logged this ticket and didn't know any better. The ticket can be closed in my opinion and as the original poster, although software crashes should always be addressed in any case.

closed 2022-07-27 by tacomarocker