Ultimaker/Cura #5613: Still locking up Ubuntu with other USB activity

FlailAway, 2019-04-11:
**Application Version**
3.6 and 4.0

**Platform**
Ubuntu 16.04 and Mint 19.1 Cinnamon

**Printer**
All, I guess

**Steps to Reproduce**
I have turned off everything I can find even remotely associated with USB in Marketplace. No exact steps, but if Cura is running and I start an Arduino app that uses the USB as a comport, after about a minute the entire system slows down. I mean slow with the mouse cursor moving 1/4" in 20-seconds. Only option is to hold down the Off-button and restart the PC.

I don't know what's in the .3mf after saving them, is there somewhere I can post the 3mf files where they are not publicly available to anyone and everyone?

Ghostkeeper, 2019-04-12:
Can't reproduce this. At least, not with the Arduino IDE. I'm running Ubuntu 18.10. It doesn't seem to slow Cura down.

You can open up the .3mf files with a zip archiver and see what's inside. There's some big XML file containing the 3D mesh data, and if you saved it as a project file it'll contain a Cura folder with some profiles. I don't expect to find anything there related to USB though.

FlailAway, 2019-04-12:
> Can't reproduce this. At least, not with the Arduino IDE. I'm running Ubuntu 18.10. It doesn't seem to slow Cura down.

Thanks for the reply, but it slows down the entire PC, essentially locking it up as it is impossible to bring up the terminal to Kill the process.

It is not just running the Arduino-IDE (1.8.6/8) it is the project using the comport between an Arduino Nano and the PC. It's only after some data transfer between them starts, it all goes wonky. If I don't have Cura running then this never happens.

The Nano and the PC are running non-stop for weeks while I use the PC for other daily stuff. Definitely something to do with Cura, but I can live with it by making sure Cura is not running if I have to restart the Nano or it's PC-app.

Just thought you would like to know.

Ghostkeeper, 2019-04-12:
Definitely sounds weird. Let's see if there's a pattern in which users encounter this.

closed 2020-06-20 by no-response[bot]