Ultimaker/Cura #5614: [4.0.0] macOs crash on Refresh Network Printers

jango-fx, 2019-04-12:
**Application Version**
4.0.0

**Platform**
macOS 10.14.3
Mac Pro (mid 2010)
3,2 GHz Quad-Core Intel Xeon
Radeon RX 480 8192 MB

Qt Version: 5.10.1
PyQt Version 5.10
OPENGL 4.1 ATI-2.4.10

**Printer**
Ultimaker 3

**Steps to Reproduce**
1. Manage Printers
2. Connect via Network
3. Refresh

**Actual Results**
CRASH

**Expected results**
Updated list of network printers
[image]

**Additional Information**

Error Traceback:

```
Traceback (most recent call last):
  File "/Applications/Ultimaker Cura.app/Contents/MacOS/plugins/plugins/UM3NetworkPrinting/src/UM3OutputDevicePlugin.py", line 241, in _onNetworkRequestFinished
    has_cluster_capable_firmware = Version(system_info["firmware"]) > self._min_cluster_version
TypeError: list indices must be integers or slices, not str
```

Log:
```
Thread 0x0000700004cba000 (most recent call first):
  File "/Users/ultimaker/build/env/4.0/inst/lib/python3.5/threading.py", line 297 in wait
  File "/Users/ultimaker/build/env/4.0/inst/lib/python3.5/site-packages/zeroconf.py", line 1569 in wait
  File "/Users/ultimaker/build/env/4.0/inst/lib/python3.5/site-packages/zeroconf.py", line 1153 in run
  File "/Users/ultimaker/build/env/4.0/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/4.0/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007000047b7000 (most recent call first):
  File "/Users/ultimaker/build/env/4.0/inst/lib/python3.5/threading.py", line 297 in wait
  File "/Users/ultimaker/build/env/4.0/inst/lib/python3.5/site-packages/zeroconf.py", line 1569 in wait
  File "/Users/ultimaker/build/env/4.0/inst/lib/python3.5/site-packages/zeroconf.py", line 1017 in run
  File "/Users/ultimaker/build/env/4.0/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/4.0/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007000042b4000 (most recent call first):
  File "/Users/ultimaker/build/env/4.0/inst/lib/python3.5/site-packages/zeroconf.py", line 946 in run
  File "/Users/ultimaker/build/env/4.0/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/4.0/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007000060c6000 (most recent call first):
  File "/Users/ultimaker/build/4.0/build/inst/lib/python3.5/site-packages/UM/Backend/Backend.py", line 169 in _storeStderrToLogThread
  File "/Users/ultimaker/build/env/4.0/inst/lib/python3.5/threading.py", line 862 in run
  File "/Users/ultimaker/build/env/4.0/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/4.0/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x0000700005bc3000 (most recent call first):
  File "/Users/ultimaker/build/4.0/build/inst/lib/python3.5/site-packages/UM/Backend/Backend.py", line 161 in _storeOutputToLogThread
  File "/Users/ultimaker/build/env/4.0/inst/lib/python3.5/threading.py", line 862 in run
  File "/Users/ultimaker/build/env/4.0/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
[... 85 lines cut ...]

Current thread 0x00000001112695c0 (most recent call first):
  File "/Users/ultimaker/build/4.0/build/inst/lib/python3.5/site-packages/cura/CrashHandler.py", line 306 in _logInfoWidget
  File "/Users/ultimaker/build/4.0/build/inst/lib/python3.5/site-packages/cura/CrashHandler.py", line 156 in _createDialog
  File "/Users/ultimaker/build/4.0/build/inst/lib/python3.5/site-packages/cura/CrashHandler.py", line 77 in __init__
  File "/Users/ultimaker/build/4.0/build/inst/bin/cura_app.py", line 98 in exceptHook
  File "/Users/ultimaker/build/4.0/build/inst/lib/python3.5/site-packages/UM/Qt/QtApplication.py", line 324 in exec_
  File "/Users/ultimaker/build/4.0/build/inst/lib/python3.5/site-packages/cura/CuraApplication.py", line 761 in run
  File "/Users/ultimaker/build/4.0/build/inst/bin/cura_app.py", line 132 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load
  File "/Users/ultimaker/build/env/4.0/inst/lib/python3.5/site-packages/cx_Freeze/initscripts/Console.py", line 21 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load
  File "/Users/ultimaker/build/env/4.0/inst/lib/python3.5/site-packages/cx_Freeze/initscripts/__startup__.py", line 12 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load

```

Ellecross, 2019-04-15:
Devs see: CL-1328 

Hey @jango-fx thanks for reporting this,
What's the version of your UM3 Firmware?

jango-fx, 2019-04-15:
oh, sry must've got lost: 5.2.8.20190320

diegopradogesto, 2019-05-03:
@ianpaschal , maybe you can take a look at this crash.