Ultimaker/Cura #5615: [4.0.0] Selection of second Extruder for skirt causes incorrect extruder selection in layer 1  for dual color models

lterrill, 2019-04-12:
**Application Version**
Cura 4.0.0

**Platform**
Mac OS 10.14.4

**Printer**
Any dual extruder printer, I used Ultimaker Original Dual Extrusion. Select two different colors of material for extruder 1 and extruder 2 to better illustrate this issue, Ultimaker Red for extruder 1, Ultimaker Yellow for extruder 2.

**Steps to Reproduce**
- Drop two models on the build plate. I used a single 20mm calibration cube and performed a 'multiple\y selected model' resulting in two models. 
- Assign extruder 1 to first model, extruder 2 to the second model.
- Select Skirt for Build Plate Adhesion, select extruder 2  as the Build Plate Adhesion Extruder. 
Note that the unexpected behavior appears unaffected by Prime tower setting

**Actual Results**
The first extruder is used to build the skirt, then the first layer of both models. The second extruder is never used on the first layer. Subsequent layers are built as expected (each extruder is used on the correct model). 
[image]
[image]

**Expected results**
Expected the model assigned to extruder 2 to have a first layer built with extruder 2. Note that I did receive the expected results when Extruder 1 is selected as the Build Adhesion Extruder and this is my current work-around. This error only occurs when Extruder 2 is selected as the Build Adhesion Extruder.
[image]

This bug appears to have been introduced in Cura version 4.

Ghostkeeper, 2019-04-12:
Not able to reproduce this. Maybe you have a project file for us?

[image]

lterrill, 2019-04-12:
Including requested project file
[Two cubes, bad first layer.3mf.zip](https://github.com/Ultimaker/Cura/files/3074055/Two.cubes.bad.first.layer.3mf.zip)

Block137, 2019-04-14:
#5550 
#5580 
Only occur when "Build plate adhesion extruder" is not extruder1
The generated G-code does use correct extruder BUT wont set temperature for it, there will be duplicate M104 T0, printer will cold extrude the skirt.

Ghostkeeper, 2019-04-15:
That project file produces a different result for me using current master.

[image]

However your bug does reproduce if I load it in 4.0. So I think that the current master introduced a new bug that caused it to not load the correct extruder from the project file, which is just hiding the bug...

It also doesn't help that due to #5550 the actual extruder that is being used is (sometimes?) hidden...

lterrill, 2019-06-01:
This issue appears to be completely resolved in Cura 4.1

closed 2019-06-01 by lterrill

Ghostkeeper, 2019-06-03:
Thanks for re-testing!